arduino/arduino-create-agent #14: Socketio Messages

matteosuppo, 2015-09-03:
We are using http://socket.io/ but only for its reconnections.

We could leverage the concept of events in order to have a clearer API.

For example:

Event from the client  | Arguments
--------  | -------------
CmdDiscover  |
CmdOpen  | {port: "", baud: ""}
CmdClose | {port: ""}
CmdSend | {port: "", data: ""}
CmdDownload | {tool: "", version: ""}
CmdTools | 
CmdInterrupt | 

Event from the agent  | Arguments
--------  | -------------
DiscoverSerial  | []array of boards
DiscoverNetwork  | []array of boards
ProgramOutput | {msg: "", type: "debug"}
ProgramSuccess | {msg: ""}
ProgramError | {msg: "", err: ""}
DownloadOutput | {msg: "", type: "debug"}
DownloadSuccess | {msg: ""}
DownloadError | {msg: "", err: ""}

facchinm, 2015-09-03:
:heart:

mastrolinux, 2015-09-03:
:tangerine: with love!