arduino/arduino-create-agent #28: Mega board not recognised on Windows IDE

hbisby, 2015-10-29:
When plugging in the Mega to Windows and using the Create IDE (on Chrome), Mega shows up as "Board at COM5" rather than the expected "Arduino/Genuino Mega 2560 at COM5". When I have tried to select the Arduino Mega as the board for the port before uploading, it still shows the "Board at COM5" in the dropdown list of connected boards as well as the Mega which is shown as disconnected. It is therefore not possible to upload a sketch to the board since the Mega is disconnected and the Board at COM5 does not know which board it is. (However it works without this problem on my Mac.)

00alis, 2015-12-03:
Same thing with the Genuino UNO

agdl, 2015-12-17:
+1 with Zero (Native USB port)

PR #43 (2016-05-18): Serial list native
Fixes #28
closed 2016-05-18 by facchinm