arduino/arduino-create-agent #45: Additional screen on Installer for Firefox cert

00alis, 2016-01-11:
[installerFirefox.pdf](https://github.com/arduino/arduino-create-agent/files/85588/installerFirefox.pdf)
Ping me if you have any question.

Here the image to add to the page:
[image]

mastrolinux, 2016-01-13:
This is going to change a bit.

Here my suggested workflow:

1. The installer works normally for every browser
2. If the user opens the getting started guide with Firefox than we will log it out
3. We expose the CA certificate and the /info API on HTTP, everything else in the daemon should be HTTPS only. 
4. We show a plain http page to the user (in the getting started guide) to allow the user accept the cert, the daemon.js script should be able to detect the http port we are listening on
5. The use accepts the certifcate, than we will redirect him to the login page and back to **https**://create.arduino.cc already redirecting to HTTPS for every request
6. We always force the user to use HTTPS, we should remove the option to avoid cert installation in the installer script

@matteosuppo @facchinm

00alis, 2016-01-13:
Glad you guys have decided for this option, it makes way more sense from UX point of view. Proceding with finalizing mockups

closed 2016-01-19 by 00alis