arduino/arduino-create-agent #48: cannot install dependency github.com/sanderhahn/gozip/patchzip

xrmx, 2016-01-12:
Trying to install dependencies but fails:
```
GOPATH=$(pwd) ./bin/godep get ./...
package github.com/sanderhahn/gozip/patchzip: cannot find package "github.com/sanderhahn/gozip/patchzip" in any of:
	/usr/lib/go/src/github.com/sanderhahn/gozip/patchzip (from $GOROOT)
	/home/rm/src/arduino-create-agent/src/github.com/sanderhahn/gozip/patchzip (from $GOPATH)
godep: exit status 1
```

facchinm, 2016-01-12:
Hi @xrmx ,
the `master` branch is out of date and will be merged soon with `dev` branch.
In the latter one all the dependencies are ok, so I strongly suggest to use that one as starting point for your hacking! :wink:

xrmx, 2016-01-12:
@facchinm good to know! :) if you are using git-flow-like branching please write it in the README.md. BTW if the master is broken what about switching the default branch in github to devel? thanks

facchinm, 2016-01-13:
`devel` is now the default branch, thanks for the tip :wink:

closed 2016-01-13 by facchinm