arduino/arduino-create-agent #49: Please add instructions on how to install dependencies

xrmx, 2016-01-12:
For go newbies (like myself) it is quite daunting how go handle package dependencies. What I've tried seems not to be sufficient:
```
export GOPATH=$(pwd)
go get github.com/tools/godep
./bin/godep get ./...
```

I got some issues like having to download dependencies with go get because godep did not find them:
```
src/github.com/onsi/gomega/ghttp/handlers.go:12:2: cannot find package "github.com/golang/protobuf/proto" in any of:
	/usr/lib/go/src/github.com/golang/protobuf/proto (from $GOROOT)
	/home/rm/src/arduino-create-agent/src/github.com/golang/protobuf/proto (from $GOPATH)
godep: exit status 1
rm@rieux:~/src/arduino-create-agent$ GOPATH=$(pwd) ./bin/godep get ./...
godep: Package (golang.org/x/sys/unix) not found
rm@rieux:~/src/arduino-create-agent$ GOPATH=$(pwd) go get golang.org/x/sys/unix
rm@rieux:~/src/arduino-create-agent$ GOPATH=$(pwd) ./bin/godep get ./...
godep: Package (github.com/getlantern/filepersist) not found
rm@rieux:~/src/arduino-create-agent$ GOPATH=$(pwd) go get github.com/getlantern/filepersist
rm@rieux:~/src/arduino-create-agent$ GOPATH=$(pwd) ./bin/godep get ./...
godep: Package (github.com/mattn/go-isatty) not found
rm@rieux:~/src/arduino-create-agent$ GOPATH=$(pwd) go get github.com/mattn/go-isatty
rm@rieux:~/src/arduino-create-agent$ GOPATH=$(pwd) ./bin/godep get ./...
godep: exec: "godep": executable file not found in $PATH
```

mastrolinux, 2016-01-13:
Looks like there are some issues on Godeps due to some broken libs. We are considering to add all the deps inside the git repo to have it always consistent.

xrmx, 2016-01-13:
@mastrolinux if it's a godeps then scratch that. Is that the corrrect way to install dependencies, if so i can do a pull request.

So after getting somehow the dependencies installed, trying ./compile_webidebridge.sh. Which fails in createZipEmbeddableFileArduino because i am missing some stuff in arduino/.

```
In createZipEmbeddableFileArduino
cp: cannot stat 'arduino/hardware': No such file or directory
cp: cannot stat 'arduino/tools_linux_arm': No such file or directory
cp: cannot stat 'arduino/boards.json': No such file or directory
cp: cannot stat 'fakecerts/*.pem': No such file or directory
cp: cannot stat 'fakecerts/*.cer': No such file or directory
mv: cannot stat 'arduino/arduino/tools*': No such file or directory
```
There workaround is easy (see below), would it  make sense to add a switch to avoid the zip creation or the missing files are really needed?
```
diff --git a/compile_webidebridge.sh b/compile_webidebridge.sh
index 82728bf..35180f6 100755
--- a/compile_webidebridge.sh
+++ b/compile_webidebridge.sh
@@ -78,9 +78,9 @@ compilePlatform()
        exit 1
        fi
        echo createZipEmbeddableFileArduino $GOOS $GOARCH $NAME
-       createZipEmbeddableFileArduino $GOOS $GOARCH $NAME
-       GOOS=$GOOS GOARCH=$GOARCH go-selfupdate $NAME $VERSION
-       rm -rf $NAME*
+       #createZipEmbeddableFileArduino $GOOS $GOARCH $NAME
+       #GOOS=$GOOS GOARCH=$GOARCH go-selfupdate $NAME $VERSION
+       #rm -rf $NAME*
 }
 
 extractVersionFromMain
```

BTW that script should probably be a Makefile.

facchinm, 2016-01-13:
I agree that the script needs some love, `createZipEmbeddableFileArduino` function is a legacy of the pre-installer version and the whole script could be simplified.
I wouldn't turn it into a Makefile because its target is simply cross compiling, since the native compilation only needs
```bash
export GOPATH=$PWD
go get
go build
```
to generate an `arduino-create-agent` binary on any platform

PR #53 (2016-01-14): README: add build instructions
Fix #49
commit 14590b35 (2016-01-14): README: add build instructions

Fix #49
closed 2016-01-14 by facchinm