arduino/arduino-create-agent #67: [Mac] white tray icon on dark scheme

00alis, 2016-02-08:
From the betatesting http://forum.arduino.cc/index.php?topic=363469.0

> The second issue was that I couldn't see that the Agent was running. My system is set to use the Dark menu bar and Dock - the ArduinoCreateAgent.app does not show an icon on the main display with the Dark menu, it does show on the non-active display menu, so I finally figured out that it was running, and the Agent took me to the http:// pages.

00alis, 2016-03-08:
Also the grey icon on Mac is really pixelated, here an updated one https://drive.google.com/a/arduino.cc/file/d/0B7BGUDKhgaDgRWo4ZkNPU2kzcW8/view?usp=sharing

00alis, 2016-07-13:
https://github.com/arduino/webide/issues/766

smellai, 2016-08-31:
http://forum.arduino.cc/index.php?topic=421586.new#new

00alis, 2016-08-31:
I know this is not a blocking issue but quite a few people are reporting it :)

smellai, 2016-09-30:
http://forum.arduino.cc/index.php?topic=426804.0

00alis, 2017-03-07:
[image]
[image]
[image]

Let me know if we need .ico and I will export the PNGs. Thanks!

closed 2017-05-29 by 00alis