arduino/arduino-create-agent #105: Downloadtools method is completely blocking

mastrolinux, 2016-06-06:
Whenever a downloadtools command is triggered the agent waits for the response and if in the meantime there are other activitities running (such as serial monitor) the agents stops reacting and we could see a disconnection in the client side.

For a full example sketch please contact @00alis

00alis, 2016-06-06:
I am just uploading this
```
void setup() {
    Serial.begin(9600);
}

void loop() {
    int val = analogRead(A0);
    Serial.print(val);
}
```

On a Zero programming port, alternating with uploading the same sketch on a Uno

00alis, 2016-06-17:
this does hasn't happened anymore, @mastrolinux are you sure it is related to downloadtools?

mastrolinux, 2016-06-17:
yes, if the connection is slow and you do not have the tools installed yet, than the call is completely blocked. You probably have the tools installed now.

hannobraun, 2018-01-30:
I haven't tried to reproduce, but I don't think this is current anymore. `downloadtool` handling is wrapped in one big goroutine: https://github.com/arduino/arduino-create-agent/blob/b128bca4e66014bed6ce0d55717d2b42540a1df6/hub.go#L169

Can this be closed?

umbynos, 2021-02-02:
As @hannobraun said, the downloadtool handling is wrapped in a go routine, this

closed 2021-02-02 by umbynos