arduino/arduino-create-agent #107: [Win] teal tray icon on blue bar, hard to see

00alis, 2016-06-17:
From the forum: http://forum.arduino.cc/index.php?topic=361376.msg2803283#msg2803283
Also maybe a slight colour change for the client side toolbar icon as here its like a faint blob on the toolbar.
And no its not a custom toolbar colour either. its windows stock.

[image]

BallscrewBob, 2016-06-22:
Just a comment. Almost all the regular Arduino icons have a solid white colour sideways 8 but these have a transparent effect. Maybe reverting to the standard logo ?

umbynos, 2021-01-18:
Fixed in https://github.com/arduino/arduino-create-agent/pull/575

closed 2021-01-18 by umbynos