arduino/arduino-create-agent #109: Prevent autostart on boot

hallolexD, 2016-07-02:
Could someone implement an option/ a setting to choose wether you want the plugin to autostart on boot. Because I don't like the fact that it just clutters my statusbar without asking. And I hate to know that software is running on my machine while I am not using the respective piece of software.

I want to be able to choose wether it should autostart or wether I will manually start the plugin whenever I want to use it.

mastrolinux, 2016-07-04:
@hallolexD can you please tell us what operating system do you use? What we could do right now is to provide instructions to disable the plugin but without implementing a specific menu, that is a lot of work to be done. If you have a working solution a PR is welcome.
Thanks for the feedback

BallscrewBob, 2016-07-07:
Its not a long process to create a shortcut, see below which worked for me.
I used "msconfig" and unchecked the create agent in the "startup" tab

Create shortcut to IDE agent.

Most taskbar programs make a shortcut in the programs list however here on Win7 X64 it didn't
I wanted to be able to shut it down and open it as I wanted.
Here is what I did.

1. Locate IDE Agent. usually in C:\Users\yourname\AppData\Roaming\ArduinoCreateAgent
Unless you have split install of windows so find your "users" folder on whatever drive.

2. Right click and select Copy.

3. Return to "dektop".

4. Right click and select "Paste Shortcut".

Now change the ICON

1. Right click your new Arduino Agent shortcut and select Properties

2. Make sure you are on the Shortcut tab and select Change Icon.

3. If you get a pop up box click OK

4. You can either choose a default icon from the selection or click the Browse

   If you want the proper Icon click Browse.

5. Browse to the path at the top (it may already select the correct path)

6. Select the "tray_iconWin" then click OK

7. Click OK again

8. You should be back on the properties page for the shortcut again and now just click Apply then OK.

Whilst this is for Win 7 the process should be similar for some other versions of windows.

mastrolinux, 2016-07-07:
I created full visual explanation in less steps here: https://github.com/arduino/arduino-create-agent#disable-autostart

closed 2016-07-07 by mastrolinux

BallscrewBob, 2016-07-07:
The pics are nicer but I included the shortcut creation and Icon selection in mine. ;)

mastrolinux, 2016-07-07:
Indeed @BallscrewBob feel free to create a Pull Request with your improvement and we will be glad to include it in the main repo.

Thanks!