arduino/arduino-create-agent #111: stuck creating desktop shortcut

elopio, 2016-08-02:
I'm using Ubuntu Yakkety 16.10.

I dowloaded ArduinoCreateAgent-1.1-linux-x64-installer.tar.gz, extracted and ran it.
Then it got stuck in 89% (Creating desktop shortcut)
[image]

This is what was printed in the terminal:
```

$ ./ArduinoCreateAgent-1.1-linux-x64-installer.run 
INFO[0000] map[]                                        
INFO[0000] Version:1.0.37                               
INFO[0000] Hostname: aeneas                             
INFO[0000] Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect. 
INFO[0000] You specified a serial port regular expression filter: usb|acm|com
 
INFO[0000] Your serial ports:                           
INFO[0000] 	{/dev/ttyUSB0     0xEA60 0x10C4 0001 false}
 
[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:	export GIN_MODE=release
 - using code:	gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET    /                         --> main.homeHandler (2 handlers)
[GIN-debug] GET    /certificate.crt          --> main.certHandler (2 handlers)
[GIN-debug] DELETE /certificate.crt          --> main.deleteCertHandler (2 handlers)
[GIN-debug] POST   /upload                   --> main.uploadHandler (2 handlers)
[GIN-debug] GET    /socket.io/               --> main.*WsServer.ServeHTTP·fm (2 handlers)
[GIN-debug] POST   /socket.io/               --> main.*WsServer.ServeHTTP·fm (2 handlers)
[GIN-debug] WS     /socket.io/               --> main.*WsServer.ServeHTTP·fm (2 handlers)
[GIN-debug] WSS    /socket.io/               --> main.*WsServer.ServeHTTP·fm (2 handlers)
[GIN-debug] GET    /info                     --> main.infoHandler (2 handlers)
[GIN-debug] POST   /killbrowser              --> main.killBrowserHandler (2 handlers)
[GIN-debug] POST   /pause                    --> main.pauseHandler (2 handlers)
[GIN-debug] POST   /update                   --> main.updateHandler (2 handlers)
INFO[0000] Inside run of serialhub                      
[GIN-debug] Listening and serving HTTP on 127.0.0.1:8991
INFO[0005] Sending err back:  We could not find the serial port /dev/ttyUSB0 that you were trying to close. 
INFO[0005] called restart                               
INFO[0005] Starting new spjs process                    
INFO[0005] exePath using osext: /home/elopio/ArduinoCreateAgent-1.1/Arduino_Create_Bridge
 
&{/home/elopio/ArduinoCreateAgent-1.1/Arduino_Create_Bridge [/home/elopio/ArduinoCreateAgent-1.1/Arduino_Create_Bridge -ls -regex usb|acm|com -gc std -hibernate] []  <nil> <nil> <nil> [] <nil> <nil> <nil> <nil> false [] [] [] [] <nil>}
FATA[0005] Exited current spjs for restart
```

elopio, 2016-08-02:
I ran it again. It got stuck in the same place, but at least the service seems to have started alright this time and I could continue on the website IDE.

```
$ ./ArduinoCreateAgent-1.1-linux-x64-installer.run 
INFO[0000] map[]                                        
INFO[0000] Version:1.0.37                               
INFO[0000] Hostname: aeneas                             
INFO[0000] Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect. 
INFO[0000] You specified a serial port regular expression filter: usb|acm|com
 
INFO[0000] Your serial ports:                           
INFO[0000] 	{/dev/ttyUSB0     0xEA60 0x10C4 0001 false}
 
[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:	export GIN_MODE=release
 - using code:	gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET    /                         --> main.homeHandler (2 handlers)
[GIN-debug] GET    /certificate.crt          --> main.certHandler (2 handlers)
[GIN-debug] DELETE /certificate.crt          --> main.deleteCertHandler (2 handlers)
[GIN-debug] POST   /upload                   --> main.uploadHandler (2 handlers)
[GIN-debug] GET    /socket.io/               --> main.*WsServer.ServeHTTP·fm (2 handlers)
[GIN-debug] POST   /socket.io/               --> main.*WsServer.ServeHTTP·fm (2 handlers)
[GIN-debug] WS     /socket.io/               --> main.*WsServer.ServeHTTP·fm (2 handlers)
[GIN-debug] WSS    /socket.io/               --> main.*WsServer.ServeHTTP·fm (2 handlers)
[GIN-debug] GET    /info                     --> main.infoHandler (2 handlers)
[GIN-debug] POST   /killbrowser              --> main.killBrowserHandler (2 handlers)
[GIN-debug] POST   /pause                    --> main.pauseHandler (2 handlers)
[GIN-debug] POST   /update                   --> main.updateHandler (2 handlers)
INFO[0000] Inside run of serialhub                      
[GIN-debug] Listening and serving HTTPS on 127.0.0.1:8991
[GIN-debug] Listening and serving HTTP on 127.0.0.1:8991
[GIN-debug] [ERROR] listen tcp 127.0.0.1:8991: bind: address already in use
INFO[0000] Error trying to bind to port: listen tcp 127.0.0.1:8991: bind: address already in use, so exiting... 
[GIN-debug] Listening and serving HTTP on 127.0.0.1:8992
http: TLS handshake error from 127.0.0.1:51818: tls: first record does not look like a TLS handshake

```

mastrolinux, 2016-08-03:
Hi,

Ubuntu 16.10 is not being released yet and it is highly experimental so it is not supported by our software. That's probably why the password dialog is not showing.

closed 2016-08-03 by mastrolinux