arduino/arduino-create-agent #114: Update udev rules during installation

mastrolinux, 2016-08-10:
To make the programming port of the Zero working we need to add the following on linux:

```
ACTION!="add|change", GOTO="openocd_rules_end"
SUBSYSTEM!="usb|tty|hidraw", GOTO="openocd_rules_end"
# CMSIS-DAP compatible adapters
ATTRS{product}=="*CMSIS-DAP*", MODE="664", GROUP="plugdev"
LABEL="openocd_rules_end"
```
I did that creating a file named `/etc/udev/rules.d/61-openocd.rules` with that content.

facchinm, 2016-08-10:
We should add a way similar to Windows' `post_install.bat` to overcome this in a standard way. Also, similar lines should be added to Linux installer so the user is directly added to `dialout` group and a warning emitted if `modem-manager` is running (the main problems for beginners in Linux right now)

mastrolinux, 2016-08-10:
I think the right group in this case is plugdev. I do agree on the post_install strategy. May I help you in some way?

facchinm, 2016-08-10:
The group depends on the distro, its existence should be checked by the install script. I believe that starting with a PR on https://github.com/arduino/ArduinoCore-samd is the right way to start :wink:

zmoog, 2021-01-26:
@facchinm are you handling this configuration step in the core now?

closed 2021-03-01 by github-actions[bot]