arduino/arduino-create-agent #116: Incorrect License.md contact details

eyak, 2016-08-15:
The license file was copied without filling in actual contact details of SW creators.
It currently reads for example:
`{signature of Ty Coon}, 1 April 1989 Ty Coon, President of Vice`

00alis, 2017-03-14:
@mastrolinux

mastrolinux, 2017-05-29:
Hi @eyak reading the licence completely it mentions:

```
You should also get your employer (if you work as a programmer) or your
school, if any, to sign a "copyright disclaimer" for the program, if
necessary.  Here is a **sample**; alter the names:
```

That means we have to add the proper header in every file (this is something we are going to do) but it is not related on the license itself because it is correct. So thank you for pointing this out but that file should not be changed. It is true we have to update other files.

zmoog, 2021-01-26:
I created a new task in our internal tracking system to add a header in every `.go` file.

Closing this issue.

closed 2021-01-26 by zmoog