arduino/arduino-create-agent #118: [Win] Installation issue

smellai, 2016-08-17:
http://forum.arduino.cc/index.php?topic=415177.0
http://forum.arduino.cc/index.php?topic=418883.new#new

smellai, 2016-09-22:
Another one with windows 7
http://forum.arduino.cc/index.php?topic=425344.new#new

facchinm, 2017-02-21:
see also https://github.com/arduino/Arduino/issues/6010

facchinm, 2017-02-21:
@anyoneInterested if you could test the development installer and report the result here it would be great. Direct link: http://downloads.arduino.cc/CreateBridge/staging/ArduinoCreateAgent-1.0-windows-installer.exe

fonix232, 2017-04-19:
I'm having the same issue on Windows 10 Pro (1607). Corporate environment, with main user not having admin rights and using a pseudo-user as SA to install stuff.

This issue happens on both the linked 1.0 development installer, and the 1.1 official.

smellai, 2018-04-19:
http://forum.arduino.cc/index.php?topic=539489.0

http://forum.arduino.cc/index.php?topic=542465.0

zmoog, 2021-01-29:
@umbynos I checked on a Windows 10 machine and and I confirm the Agent 1.2 installer requires admin privileges (the app icon has the blue and yellow shield and the User Account Control (UAC) kicks in when you run it).

per1234, 2023-11-18:
Closing as duplicate of https://github.com/arduino/arduino-create-agent/issues/673. Even though this one was submitted much earlier, the other contains more information so I give it preference.

closed 2023-11-18 by per1234