arduino/arduino-create-agent #123: Documentation for the HTTP(S) interface of Create Agent

joachimhs, 2016-08-22:
Hello, 

I have noticed that the Arduino Create Agent offers HTTP(S) and WebSocket endpoints for talking to the Arduino and to upload sketches. I am developing a block-based Arduino system in order to help teach Arduino and electronics to kids via my code club and makerspace (blockuino.no), and I am investigating the possibility of including functionality to upload sketches directly from blockuino.no to the Arduino via these HTTP(S) or WebSocket endpoints. 

So far, though I am having little luck. I can generate a Curl copy of the request that the Arduino Create website creates when uploading the Hex file, but I am unable to replicate this via HTTP(S) at all. 

Is there an example or a guide of sorts that I can look at in order to complete this integration? 

For completeness I am attaching the full HTTP Request that I am trying to send to the Arduino. The result is a "200 OK" response, but nothing really happens on the Arduino itself. 

HTTP Request: https://gist.github.com/joachimhs/152a21399f51be43ce9f84ff0f5379cd

mastrolinux, 2016-08-29:
Hi,

First of all you have to use the https port in the agent to have it working with https. It usually is at the next port than the http. So if the http port is at localhost:8990, than the https is at localhost:8991 (https://localhost:8991/) 

The other important thing to do for security reason in to add the signatureKey of your server as you can read here: https://github.com/arduino/arduino-create-agent/blob/66ed59ad4bd139bdbc92974b31e8cf196861f1f0/main.go#L47

it has to be a pgp signature replacing the carriage return with the `\n` value.

Thank you for testing this feature and let me know if you need more info.

mastrolinux, 2016-10-07:
Some more info to create your own key on the server side:
```
$ openssl genrsa -out commandline 2048
$ openssl rsa -pubout -in commandline -out public.key
$ sed ':a;N;$!ba;s/\n/\\n/g' public.key
$ cat public.key
```

closed 2016-11-29 by mastrolinux