arduino/arduino-create-agent #129: Arch Linux Chromium Issue

NonaSuomy, 2016-11-08:
Just to note this is all working in FireFox 49.0.2 after I found out I had to re-run the getting started process to accept a cert for firefox...

I was first trying to get the plugin working with Chrome which I had no luck on, so then I tried firefox. I didn't realize that you had to run the plugin interface again in the browser (as I already had the bridge installed) to get this cert nfo to pop up.  Just thought it would be universal for the platform you are working with, might want to handle this differently as I didn't know what to do until I found these issue posts below then it dawned on me. That I had to then re-run the getting started stuff and then it worked great in FF. :+1: Maybe detect that they don't have the cert and point them to the getting started guide.

https://github.com/arduino/arduino-create-agent/issues/45
https://github.com/puma/puma-dev/issues/47#issuecomment-259231445

Back to Chromium Version 54.0.2840.71 (64-bit)

```
Something went wrong

We are not able to establish a secure (HTTPS) connection between the Plugin and the Arduino Web Editor. You can try to restart your browser once more, or Download the Installer and run it again.
Please Write us on the Forum so we can help debugging this isse.
```

Minor note there is a typo in the word issue at the end of the sentence there. ^

Console:

```
ArduinoCreateAgent]$ ./Arduino_Create_Bridge 
INFO[0000] map[avrdude:/home/username/.arduino-create/avrdude/6.3.0-arduino6 avrdude-6.3.0-arduino6:/home/username/.arduino-create/avrdude/6.3.0-arduino6] 
INFO[0000] Version:1.0.46                               
INFO[0000] Hostname: systemname                             
INFO[0000] Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect. 
INFO[0000] You specified a serial port regular expression filter: usb|acm|com
 
INFO[0000] Your serial ports:                           
INFO[0000] 	{/dev/ttyACM0     0x8036 0x2341  false}
    
INFO[0000] 	{/dev/ttyUSB0     0x6001 0x0403 A6004mMA false}
 
[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:	export GIN_MODE=release
 - using code:	gin.SetMode(gin.ReleaseMode)

INFO[0000] Inside run of serialhub                      
[GIN-debug] GET   /                         --> main.homeHandler (2 handlers)
[GIN-debug] GET   /certificate.crt          --> main.certHandler (2 handlers)
[GIN-debug] DELETE /certificate.crt          --> main.deleteCertHandler (2 handlers)
[GIN-debug] POST  /upload                   --> main.uploadHandler (2 handlers)
[GIN-debug] GET   /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] POST  /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] WS    /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[GIN-debug] WSS   /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] GET   /info                     --> main.infoHandler (2 handlers)
[GIN-debug] POST  /killbrowser              --> main.killBrowserHandler (2 handlers)
[GIN-debug] POST  /pause                    --> main.pauseHandler (2 handlers)
[GIN-debug] POST  /update                   --> main.updateHandler (2 handlers)
[GIN-debug] Listening and serving HTTP on 127.0.0.1:8991
[GIN-debug] Listening and serving HTTPS on 127.0.0.1:8991
[GIN-debug] [ERROR] listen tcp 127.0.0.1:8991: bind: address already in use
INFO[0000] Error trying to bind to port: listen tcp 127.0.0.1:8991: bind: address already in use, so exiting... 
[GIN-debug] Listening and serving HTTPS on 127.0.0.1:8992

(Arduino_Create_Bridge:14991): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
```

```
ArduinoCreateAgent-1.1]$ ./Arduino_Create_Bridge 
INFO[0000] map[avrdude:/home/username/.arduino-create/avrdude/6.3.0-arduino6 avrdude-6.3.0-arduino6:/home/username/.arduino-create/avrdude/6.3.0-arduino6] 
INFO[0000] Version:1.0.38                               
INFO[0000] Hostname: systemname                             
INFO[0000] Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect. 
[... 114 lines cut ...]

```
Restart Chrome

We know this is annoying, but to update the Certificates settings you need to restart Chrome.
If you don't restart the browser, the Arduino Web Editor won't be able to communicate with the plugin. The browser will reopen just this tab.
Any issue or question? Write us on the Forum.
```
Click Restart:

Loading Arduino Create...

CLI:
```
http2: server: error reading preface from client 127.0.0.1:53542: read tcp 127.0.0.1:8992->127.0.0.1:53542: read: connection reset by peer
```

WebGUI:

```
Something went wrong

We are not able to establish a secure (HTTPS) connection between the Plugin and the Arduino Web Editor. You can try to restart your browser once more, or Download the Installer and run it again.
Please Write us on the Forum so we can help debugging this isse.
```

facchinm, 2016-11-11:
Hi @NonaSuomy ,
the fix here is not simple, but I can try to give you some hints (being a fellow Archer I can relate :smile: )

First of all, the complicated certificate dance is necessary because otherwise the Create webpage would contain both secure and insecure sources. So the local webserver (and websocket) must be reached via https (not selfsigned). 

To accomplish this, the installer calls `arduino-create-agent` with `--generateCert` during the installation. This generates a couple of certs+keys (one root and one derived). The root one gets copied to the $OSDependant keyring and then its key gets removed (so it can't be used to sign other intermediate certificates).

The problem here is that newer Linux distro have different (default) keyring location and interaction methods. Right now, we use the following snippet 
```bash
if cmd_exists update-ca-certificates; then
    ca_path=/usr/local/share/ca-certificates/
    ca_update_cmd=update-ca-certificates
else
if cmd_exists update-ca-trust; then
    ca_path=/usr/share/ca-certificates/trust-source/anchors/
    ca_update_cmd=update-ca-trust
else
```

then we copy `createAgentLocal.crt` into `$ca_path` and call `$ca_update_cmd`. You can try executing this manually, if it fails you will get the root cause of the problems.

Firefox is still different, since it doesn't use the system keyring but its own. So, if during the installation Firefox is detected, we land you on a specially crafted page to accept the certificate. If you run the "getting started" on another browser the Firefox popup will never be displayed, so you must re-execute the installer within FF.

Last but not least, the agent autoupdates itself in an unattended way, so if you see the program "crashing" it probably received an update request. The installer always contains the latest version available.

Hope it helps, I'm currently running `1.0.184-b199b1a` on Arch and it works just good.

NonaSuomy, 2016-11-20:
Hi facchinm,

Nice to meet another Arch user :+1: 

ca_path=/usr/share/ca-certificates/trust-source/anchors/ exists.

I Checked this path and it did not contain createAgentLocal.crt

This is the ArduinoCreateAgent folder content:

```
ArduinoCreateAgent-1.1]$ ls
Arduino_Create_Bridge
ca.cert.cer
ca.cert.pem
ca.key.pem
cert.cer
certmgr.sh
cert.pem
config.ini
InstallerIcon.png
key.pem
uninstall
'Uninstall arduino-create-agent.desktop'
uninstall.dat
```

Seems this createAgentLocal.crt doesn't exist or did it do something to it after running...

```
ArduinoCreateAgent-1.1]$ sudo ./Arduino_Create_Bridge --generateCert
INFO[0000] written ca.key.pem                           
INFO[0000] written ca.cert.pem                          
INFO[0000] written ca.cert.cer                          
INFO[0000] written key.pem                              
INFO[0000] written cert.pem                             
INFO[0000] written cert.cer
```

```
ArduinoCreateAgent-1.1]$ sudo ./certmgr.sh 
pkexec
/usr/bin/cp: missing destination file operand after '/usr/share/ca-certificates/trust-source/anchors//createAgentLocal.crt'
Try '/usr/bin/cp --help' for more information.
```

Are you supposed to rename one of those files above to createAgentLocal.crt?

I tried to compile the latest github source to maybe get the version you have, as downloading from the site only gets me the same Version:1.0.46.

I did the steps

```
export GOPATH=$PWD
go get
go build
```

(I commented out the OSX and Windows build at the end of the script)

seemed like it compiled but not sure where it went.

I then noticed this link, that I didn't before:

http://downloads.arduino.cc/CreateBridge/staging/ArduinoCreateAgent-1.0-linux-x64-installer.run

I ran the above and it gave me 1.0.188, which still had the same issue so I think the main issue is that createAgentLocal.crt is missing?

```
ArduinoCreateAgent-1.0]$ sudo ./Arduino_Create_Bridge
http2: server: error reading preface from client 127.0.0.1:48146: read tcp 127.0.0.1:8994->127.0.0.1:48146: read: connection reset by peer
```

Thanks for the help, I feel like were getting closer to the culprit.

NonaSuomy, 2016-11-21:
I found this post from you in the forums: https://forum.arduino.cc/index.php?topic=411290.0

It says "* copy ca.cert.cer to your local certificate pool and refresh the index (maybe use this script to understand where it needs to be copied)".

So I did:

```
sudo cp ca.cert.cer /usr/share/ca-certificates/trust-source/anchors/
sudo update-ca-trust
```

Then I restarted Arduino_Create_Bridge <= I think you should change this to the word "Agent" instead of "Bridge" as it's a little confusing to have two different names for something. Then I restarted Chromium and all of a sudden I see "Arduino Leonardo" Connected! w00t.

Finally thanks for your help, they may want to fix that:
```
/usr/bin/cp: missing destination file operand after '/usr/share/ca-certificates/trust-source/anchors//createAgentLocal.crt'
```

Instead of ```'ca.cert.cer'``` .

Thank again!

closed 2017-02-13 by facchinm