arduino/arduino-create-agent #131: Error Downloading libraries from Arduino online editor

alieslam, 2016-12-19:
As stated in the attached picture, i couldn't manage to download libraries from Arduino's online editor. I have tried to download libraries from "Arduino" and not just "Favorites" but seems working for some libraries and others don't.  

[image]

smellai, 2016-12-19:
Hi @alieslam , you are right I'm opening an issue in our create editor repository, if you find any other issue in the editor you can write us here http://forum.arduino.cc/index.php?board=101.0
Thanks!

closed 2016-12-19 by smellai

alieslam, 2016-12-19:
Hi @smellai ,Thanks for your quick response, I do have another issues so i will be posting them on the forum :).

00alis, 2016-12-19:
Thanks a lot!

smellai, 2016-12-19:
It's a bug with the libraries that have subfolders in the examples. We are going to fix it as soon as possible