arduino/arduino-create-agent #133: [ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use

nekhbet, 2017-01-06:
On Ubuntu 16, with a fresh install I get: 
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use

Also when running "lsof -i udp:5353" I get nothing ?!

facchinm, 2017-01-09:
Hi @nekhbet ,
that error refers to mdns discovery on ipv6, which is something we don't yet support but it's bundled with the library in use. It should behave as a warning (ie. the agent shouldn't exit). Is it correct?

nekhbet, 2017-01-15:
You are right, it doesn't exit but also the browser can't connect to it either.

facchinm, 2017-01-16:
To establish the connection with the browser, a self hosted SSL certificate is needed. We provide it during the installation but we need to refine the procedure **a lot** . If you could, in this order:
- kill the running plugin (via the tray icon or simply with `killall Arduino_Create_Bridge` )
- restart it from the commandline
- paste here the output when you try to open the https://create.arduino.cc/editor website

it would help us a lot in solving the issue, thanks!

nekhbet, 2017-02-07:
Sorry for the delay, I missed this notification.

So .. I did the first step: "sudo killall Arduino_Create_Bridge")
then re-started it ("sudo /opt/ArduinoCreateAgent-1.1/Arduino_Create_Bridge")

and I got this on the command line: 

```
INFO[0000] map[]                                        
INFO[0000] Version:1.0.46                               
INFO[0000] Hostname: unknown                            
INFO[0000] Garbage collection is on using Standard mode, meaning we just let Golang determine when to garbage collect. 
INFO[0000] You specified a serial port regular expression filter: usb|acm|com
 
INFO[0000] Your serial ports:                           
INFO[0000] 	There are no serial ports to list.          
[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:	export GIN_MODE=release
 - using code:	gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET   /                         --> main.homeHandler (2 handlers)
[GIN-debug] GET   /certificate.crt          --> main.certHandler (2 handlers)
[GIN-debug] DELETE /certificate.crt          --> main.deleteCertHandler (2 handlers)
[GIN-debug] POST  /upload                   --> main.uploadHandler (2 handlers)
[GIN-debug] GET   /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] POST  /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] WS    /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] WSS   /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] GET   /info                     --> main.infoHandler (2 handlers)
[GIN-debug] POST  /killbrowser              --> main.killBrowserHandler (2 handlers)
[GIN-debug] POST  /pause                    --> main.pauseHandler (2 handlers)
[GIN-debug] POST  /update                   --> main.updateHandler (2 handlers)
[GIN-debug] Listening and serving HTTP on 127.0.0.1:8991
INFO[0000] Inside run of serialhub                      
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[GIN-debug] Listening and serving HTTPS on 127.0.0.1:8991
[GIN-debug] [ERROR] listen tcp 127.0.0.1:8991: bind: address already in use
INFO[0000] Error trying to bind to port: listen tcp 127.0.0.1:8991: bind: address already in use, so exiting... 
[GIN-debug] Listening and serving HTTPS on 127.0.0.1:8992

(Arduino_Create_Bridge:804): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:23:18: not a number

(Arduino_Create_Bridge:804): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:23:18: Expected a string.

(Arduino_Create_Bridge:804): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:832:10: not a number

(Arduino_Create_Bridge:804): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:832:24: Using Pango syntax for the font: style property is deprecated; please use CSS syntax

(Arduino_Create_Bridge:804): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:1985:14: not a number

(Arduino_Create_Bridge:804): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:1985:14: Expected a string.

(Arduino_Create_Bridge:804): Gtk-WARNING **: Theme parsing error: nautilus.css:55:71: Using one color stop with linear-gradient() is deprecated.

(Arduino_Create_Bridge:804): Gtk-WARNING **: Theme parsing error: nautilus.css:56:71: Using one color stop with linear-gradient() is deprecated.
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
```

on the browser .. it asks me to restart Chrome, I do this but the process itself fails so i manually restart it but the process is again repeated (asking me again to restart for the install to complete).

Thanks.

nekhbet, 2017-02-07:
Also .. accessing 127.0.0.1:8991 works on HTTP but on HTTPS: 

```
This site can’t provide a secure connection

127.0.0.1 sent an invalid response.
ERR_SSL_PROTOCOL_ERROR
```

nekhbet, 2017-02-07:
Ha .. what seems to have solved it was this: access from browser: https://127.0.0.1:8992/ and manually accept the self-signed certificate.

facchinm, 2017-02-07:
HTTPS should be reachable on :8992, and the Create website should use that port. From the logs it seems that the certificates are ok, anyway try to access https://localhost:8992 and check if the certificate is ok

facchinm, 2017-02-07:
Great, so the problem was something related with the certificates not being installed system-wide. Are you on Ubuntu 16.04 64bit ? So I can check if the installation works on a VM

nekhbet, 2017-02-07:
yes, that's the OS

```
cat /proc/version
Linux version 4.8.0-34-generic (buildd@lcy01-21) (gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12) ) #36-Ubuntu SMP Wed Dec 21 17:24:18 UTC 2016
```

facchinm, 2017-03-21:
@nekhbet we solved the certificate installation with the latest installer, would you mind testing again so I can close the issue? Thanks!

nekhbet, 2017-04-02:
Works now. Thank you!

However I noticed another bug "Go to Arduino Create" doesn't open the default system browser but Firefox.
[image]

facchinm, 2017-04-03:
We use https://github.com/skratchdot/open-golang/ to open the link with the default browser, but it looks like `xdg-open` and Ubuntu's "default application"  selector are not exactly the same thing. 
I'll take a look at this, in the meantime I'm closing the issue as solved (for the certificate part)

closed 2017-04-03 by facchinm