arduino/arduino-create-agent #139: Uninstalling

troywweber7, 2017-02-14:
I am not quite sure where I downloaded from, but in my downloads folder I have `ArduinoCreateAgent-1.1-linux-x64-installer.tar.gz` and `ArduinoCreateAgent-1.1-linux-x64-installer.run`.

If I `ArduinoCreateAgent-1.1-linux-x64-installer.run --help`, I get

```
arduino-create-agent 1.1
Usage:

 --help                                      Display the list of valid options

 --version                                   Display product information

 --unattendedmodeui <unattendedmodeui>       Unattended Mode UI
                                             Default: none
                                             Allowed: none minimal minimalWithDialogs

 --optionfile <optionfile>                   Installation option file
                                             Default: 

 --debuglevel <debuglevel>                   Debug information level of verbosity
                                             Default: 2
                                             Allowed: 0 1 2 3 4

 --mode <mode>                               Installation mode
                                             Default: gtk
                                             Allowed: gtk xwindow text unattended

 --debugtrace <debugtrace>                   Debug filename
                                             Default: 

 --installer-language <installer-language>   Language selection
                                             Default: en
                                             Allowed: sq ar es_AR az eu pt_BR bg ca hr cs da nl en et fi fr de el he hu id it ja kk ko lv lt no fa pl pt ro ru sr zh_CN sk sl es sv th zh_TW tr tk va vi cy

 --prefix <prefix>                           Installation Directory
                                             Default: /home/troyw/ArduinoCreateAgent-1.1
```

Do I only need to remove the directory it installed to, or are there other files I should remove to uninstall it?

facchinm, 2017-02-14:
Hi @troywweber7 ,
the installer creates its own uninstaller in `$HOME/ArduinoCreateAgent-1.x/`, simply running `Uninstall Arduino_Create_Bridge.desktop` into that folder should do the trick

closed 2017-03-14 by 00alis

Kurio71, 2017-03-27:
Trying to install, arduino-create-agent 1.1 , silently with the switches --unattendedmodeui minimalWithDialogs --mode win32 unattended. The installer doesne't like unattended. Anyone got this working?

facchinm, 2017-03-27:
Hi @Kurio71 ,
the installer is build with [Installbuilder](https://installbuilder.bitrock.com/), so the right switch for unattended should be `--mode unattended`  (from the docs, I didn't actually tried it)

Kurio71, 2017-03-27:
Thanks, I'll give it a go

cchaglund, 2019-12-18:
Can't uninstall on Catalina v10.15.2. I double click the uninstall.app and the icon animates as if it's about to be opened, then nothing happens.