arduino/arduino-create-agent #144: Need this module modified for Chromebook compatibility...

randmor, 2017-02-23:
Lots of schools these days are using Chomebooks and they also want to do Arduino classes or labs and are unable to do so because while the Arduino Web Editor works on Chrome, the create agent (which I assume to be the piece of software that is used to upload the object code to the USB connected Arduino MCU board ala AVR Dude. On previous 3rd party attempts at filling this void they used a piece of software called AVR Chick. I am not sure, but maybe this code (AVR Chick) could be used instead? In any case, a lot of folks are looking for alternatives to Arduino simply because its not available to run on Chromebooks. Other MCU boards are, including BBC Micro:Bit and CodeBug (maybe others?)

randmor, 2017-02-23:
AVRChick is described as a NodeJS library for flashing compiled sketch files to Arduino microcontroller boards. You can likely find the source for AVRChick at this GitHub URL:
https://github.com/noopkat/avrgirl-arduino

Wish I were a better programmer and could do this work myself.

randmor, 2017-02-23:
Shows you how much I know about GitHub... I accidently closed my issue

facchinm, 2017-02-24:
Hi @randmor ,
the agent as it is can't be installed on Chromebooks (for obvious reasons, since it's a desktop app). However, we are working heavily to support Chromebooks, so keep tuned for updates :slightly_smiling_face:

lila, 2017-07-08:
how is the chromebook version coming along?  i too would love the option of using the arduino web editor and create agent for chromebooks (without resorting to crouton).  happy to be a beta/alpha tester...

randmor, 2017-07-08:
I have not heard anything new. I am also needing the capability of running the Arduino IDE on Chromebook as that is what our school district has designated as the officially supported machine. What have I tried to get around this? I have installed Ubuntu Linux on a Chromebook and ran the Linux version of Arduino and this works fine, but teaching Linux to elementary and middle school kids is a bit of a stretch to say the least. But the Chromebook makes a nice and inexpensive Linux laptop that is pretty price competitive with the Raspberry Pi once you add in all the costs for the monitor, keyboard, mouse, memory card, etc. What you get with the Chromebook-turned-Linux machine is everything is nicely integrated together, and the end result is a much faster Linux machine.
The other option was to go an buy 10 refurbished Windows 7 Dell Latitude Model D630 PCs for like $120 per unit and use them (and support them) independently of the school's IS department. I'm waiting for the other foot to drop, but so far it hasn't. And it solves the problem nicely.

00alis, 2017-07-10:
Hi @randmor and @lila,
Arduino Create is currently available on Chrome OS, the beta testing is finally over.
You can install it via the Google Store at https://chrome.google.com/webstore/detail/arduino-create/dcgicpihgkmccjigalccipmjlnjopdfe.

You can report any issue you might encounter at https://forum.arduino.cc/index.php?board=112.0

closed 2017-07-10 by 00alis