arduino/arduino-create-agent #147: Web Editor plugin not working: cannot upload to MKR 1000

oliver-gilmour-twi, 2017-03-17:
Hi,

I'm using Windows7, 64 bit. The computer is a company one, so all internet traffic goes through HTTP proxy.
I'm also using Firefox 47.0.2.

I've installed the plugin, and that seems to be working (I see the green and black Arduino icon on the system tray), and I go to the Arduino Web Editor, but I keep getting the message "No Plugin Connection. Uploading is disabled until you reconnect".

I've installed the driver (because I've also installed desktop IDE - there's another bug there because of proxy implementation, but that's another story) and the system does recognise the Genuino MKR 1000 board. I just can't program it.

facchinm, 2017-03-21:
Hi @oliver-gilmour-twi ,
could you test by whitelisting `localhost` in Firefox proxy setting? Also, it could be needed to inject the environment variable `HTTP_PROXY` before starting the agent.
I'm testing with a local `squid` installation and things seem to start working, but I need some insight from someone behind a real proxy to make sure things are working correctly.

saniales, 2017-08-24:
we will add it in the CLI config file

mastrolinux, 2017-11-15:
we should put proxy settings vars in the config file of the agent itself if possible.

mastrolinux, 2018-01-12:
We are now able to reproduce, detailed explanation here:
http://nanxiao.me/en/set-proxy-when-executing-go-get-command/

hannobraun, 2018-01-22:
I've started to look into this. I managed to reproduce the issue locally, and am now working on confirming the workaround that @mastrolinux linked.

hannobraun, 2018-01-23:
Since last posting here, I've gained a better understanding of what's happening, and I'm no longer sure about what was previously said in this issue.

Fist, the editor communicates with the agent by making requests to `localhost`. I don't see how this could ever work over a proxy, since `localhost` means something different to the proxy than it means to the web editor. Changing the proxy settings to bypass `localhost`, as suggested by @facchinm, seems to be the only solution. (Unless, of course, the proxy is running on the same machine. But then things depend entirely on how the configuration of the proxy, which is outside of our control anyway.)

Second, the article that @mastrolinux linked is about setting environment variables that configure Go programs (that use net/http) to use a proxy for making requests. I don't see how this applies here, for two reasons:
1. It won't change anything about the requests that the browser is making to `localhost`, which seem to be the problem here.
2. If there is a system-wide proxy, as the issue description indicates, outgoing HTTP requests from the agent presumably already go through a proxy. I don't understand what setting the configuration explicitely would change about that.

It is entirely possible that I'm missing something, as I'm not very familiar with arduino-create-agent yet, nor do I have much practical experience with HTTP proxies. If I'm wrong about my analysis, or am misunderstanding something about this issue, any guidance would be greatly appreciated.

mastrolinux, 2018-01-23:
@hannobraun the fact is point 2. is not correct.

On Windows environment variables are not available if the process runs as daemon, that's the issue, there is not env defined at all so there are no https_proxy and http_proxy variables available and we have to export them in the config.ini file.

Point 1. is not correct too: the agent tries to download some tools available at https://downloads.arduino.cc and it fails because the agent is not configured to be behind a proxy. This is completely unrelated about localhost.

hannobraun, 2018-01-23:
@mastrolinux Thank you for your reply.

> the fact is point 2. is not correct.
> 
> On Windows environment variables are not available if the process runs as daemon, that's the issue, there is not env defined at all so there are no https_proxy and http_proxy variables available and we have to export them in the config.ini file.

Understood, thank you.

> Point 1. is not correct too: the agent tries to download some tools available at https://downloads.arduino.cc and it fails because the agent is not configured to be behind a proxy. This is completely unrelated about localhost.

Okay, I think I'm missing some context here. According to the issue description the connection to the agent doesn't seem to be working at all ("No Plugin Connection. Uploading is disabled until you reconnect"). There wasn't any mention of trying to download tools.

Is your request to add proxy configuration to config.ini related to the original report, or is it something separate? I will add those options, no problem. At this point I'm just confused about what exactly this issue is about :-)

mastrolinux, 2018-01-25:
The issue is the plugin does not work behind a proxy on windows while downloading other tools because of how Windows exports variables for deamons. We have to add a proxy config in the .ini file.

hannobraun, 2018-01-25:
> The issue is the plugin does not work behind a proxy on windows while downloading other tools because of how Windows exports variables for deamons. We have to add a proxy config in the .ini file.

Thank you, I understood that this is what you want after reading your previous comment, and I'm already working on it (still figuring out #199). I don't understand if this is related to the issue originally reported by @oliver-gilmour-twi, which is were my confusion comes from.

smellai, 2018-07-02:
Added instructions to set up proxy configuration
https://github.com/arduino/arduino-create-agent#when-behind-a-proxy

closed 2018-07-02 by smellai

jtauscher, 2019-02-01:
For me it doesnt work, adding a Proxy. If I change the Agent to use the Proxy.ini it just closes the Plugin and thats it.
I just added following line at the end and it looks like this :
**

```
name = Proxy Enabled
gc = std  
hostname = unknown-hostname  # Override the hostname we get from the OS
regex = usb|acm|com  # Regular expression to filter serial port list
v = true  # show debug logging
appName = CreateBridge
updateUrl = http://downloads.arduino.cc/
origins = http://local.arduino.cc:8000
http_proxy="http://name:pwd@10.102.4.91:8080"
```

**