arduino/arduino-create-agent #149: binary sketch uploading POST request

elhele, 2017-05-03:
Hello,

I was trying to test an example of the POST request that you provide using Firefox REST client, but I got a message that the signature is wrong. Should I have generated another one? I don't have the private key to generate the signature. Or should I create a key pair by myself, sign the string and manually change the public key in the main.go of the program? In README it is said that "public key contained in the config.ini", but I don't see any key there and also in conn.go it seems that they key is taken not from the config.ini, but from a variable signatureKey in main.go (but here I can be wrong -- I'm not really familiar with go).

The thing is that we are thinking about using the plugin in our project (https://lab.open-roberta.org/) for uploading the programs to Arduino and for this I need to understand how to generate the correct call.

Thank you in advance,
Evgeniya

closed 2017-05-16 by elhele

facchinm, 2017-05-16:
Hi @elhele ,
sorry for the late response but I totally missed this issue :slightly_smiling_face: 
In go, all variables marked as `flag.*` are evaluated by `flag.Parse()` , even if not explicitely stated. 
Adding an entry on `config.ini` with `signatureKey = -----BEGIN PUBLIC KEY-----\nMIIBIjANBgkqhkiG9...` should do the trick

elhele, 2017-05-18:
Hello. Thank you very much for the reply, I've already got it, I just used a wrong encoding for signing :)
Now it seems to be working fine and we've almost integrated it into our project, probably it'll be included into the next release.