arduino/arduino-create-agent #150: CORS Error when trying to connect a webpage in chrome to arduino-create-agent

arthursw, 2017-05-16:
I would like to connect a webpage from chrome to *arduino-create-agent*, but I have the following error:

```XMLHttpRequest cannot load http://localhost:8991/info. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'null' is therefore not allowed access.```

Here is the javascript code from my html page which test the possible ports until I get a reply (in typescript):

```typescript
let portNumber = 8990

let connectionSuccess = function(data: any, textStatus: string, jqXHR: any) {
	console.log( "connection to arduino-create-agent success" )
	if(data.hasOwnProperty('ws')) {
		this.openWebsocketConnection(data.ws)
	}
}

let connectionError = function(jqXHR: any, textStatus: string, errorThrown: any) {
	portNumber++
	if(portNumber > 9000) {
		console.log( "Error: impossible to connect to arduino-create-agent" )
		return
	}
	connectArduinoCreateAgent()
}

let connectArduinoCreateAgent = function() {
	return $.ajax( { url: "http://localhost:" + portNumber + "/info" } ).done(connectionSuccess).fail(connectionError)
}

connectArduinoCreateAgent()

```
and then to connect to websocket once I get the proper port:

```typescript
	openWebsocketConnection(websocketPort: string) {
		this.socket = new WebSocket(websocketPort)

		this.socket.addEventListener('open', (event) => {
			this.socket.send('list')
		})

		this.socket.addEventListener('message', (event: any) => {
			console.log('Message from server', event.data)
		})

	}
```

How can I get rid of this CORS error? Is it even possible?

matteosuppo, 2017-05-16:
In the Agent installation folder there's a `config.ini`.

You can modify the parameter 'origins' and restart your agent

arthursw, 2017-05-16:
Thanks! This worked :)

However I have [a websocket issue](https://github.com/arduino/arduino-create-agent/issues/151) now. I had this problem before but I thought it was also due to the cross origin error.

closed 2017-05-16 by arthursw