arduino/arduino-create-agent #151: Error during WebSocket handshake: Unexpected response code: 200

arthursw, 2017-05-16:
I have the following error when I try to connect my webpage (from chrome) to *arduino-create-agent* through websockets:
 
```WebSocket connection to 'ws://localhost:8991/' failed: Error during WebSocket handshake: Unexpected response code: 200```

Here is the code that I use: 

```typescript
let socket = new WebSocket('ws://localhost:8991/');
```

matteosuppo, 2017-05-16:
The agent uses socket.io (https://socket.io/docs/) so you'll probably need to open the websocket to wss://localhost:8992/socket.io/ or just use socket.io

arthursw, 2017-05-16:
Yes, it works with `socketio`. I don't understand why it does with native WebSockets but it works.

closed 2017-05-16 by arthursw

viettranme, 2018-08-17:
For working with native WebSockets, try adding _/websocket_ to the end of your websocket's URL, for example, like: 

`ws://localhost:8991/websocket`