arduino/arduino-create-agent #152: Unable to see ports for multiple boards

gguy1, 2017-05-28:
I don't see multiple ports when I connect multiple 101's to my machine.

I wanted to test using the BLE libraries one as central and another as peripheral. However I could not see both boards on my port list using the web editor.

- I am trying to do this in a chrome browser

- Ubuntu 16.04 OS

00alis, 2017-05-29:
Would you send us a screenshot of what you see in the board and ports dropdown as in the one below? Thanks.
[image]

facchinm, 2017-05-29:
Hi @gguy1 , one board is probably stuck in DFU mode (due to a bad update or some other reason). Simply attach one board at a time and flash them with a Blink. Once you make sure that both the boards expose the serial interface independently you can retest with both attached and they should both appear. If it doesn't happen, could you provide the debug output of your agent (right click on the tray, open debug console)? Thanks!

gguy1, 2017-05-29:
I was able to see both of them trying the steps mentioned.
Thank You,
Gavin Guy

facchinm, 2017-05-29:
Closing as solved

closed 2017-05-29 by facchinm