arduino/arduino-create-agent #156: Real and virtual COM ports are not listed

zmechanic, 2017-06-22:
It seems that there is a restriction in seriallist.go to list only USB CDC ports. All other non-USB ports are ignored. I removed this restriction in my local repo, and got ports displayed in Arduino Web IDE, but upload is not working for them. After build web front-end is stuck and upload button forever displays BUSY.

facchinm, 2017-06-23:
Hi @zmechanic .
I believe your modification regards [this](https://github.com/arduino/arduino-create-agent/blob/devel/seriallist.go#L45) line. It shouldn't have any side effect, but please remember that some boards need a particular reset sequence which involves the extra UART signals like DTS, RTS and so on. Which board are you targeting?

zmechanic, 2017-06-23:
Hi @facchinm .
Yes, nearly, just about few lines [above](https://github.com/arduino/arduino-create-agent/blob/devel/seriallist.go#L40).

What I have done is that I added a fake entry into the list (e.g. COM8). This fake COM8 port successfully turned up in Arduino Web IDE. The IDE allows me to assign a board (Arduino Nano) and MCU (ATmega328) to it. Then I'm able to click Upload button in Web IDE. At that moment it compiles code in the cloud, and prints stats about used memory space in the console output on the web page. Unfortunately, nothing happens after that. It just stuck in this state with button displaying BUSY.

From my understanding of arduino-create-agent code, Web IDE must do POST to /upload page self-hosted by arduino-create-agent. This doesn't happen for some reason. I confirmed it with Fiddler and with Chrome network activity view.

I wonder what happens there, and why it decides not to call /upload.

facchinm, 2017-06-23:
Mmmmh, can you post the verbose output (you can enable it from the preferences) and the relevant debug output (right click on the tray icon, "open debug console")? Thanks!

zmechanic, 2017-06-23:
It looks like it was my fault (sort of) in the end. While faking a port I didn't assign correct VID and PID, considering that only USB CDC ports are allowed, which should always report it back.

So, I grabbed VID and PID from real board and assigned them to my fake port. Everything works without a hitch now. I think there is something inside, which does cross reference between VID and PID reported by hardware and upload process. So when those values are absolute nonsense the upload process doesn't start, even if target board was selected from dropdown in the web UI.

Not sure if this need to be closed now, as this indicates some sort of bug, I guess.

facchinm, 2017-06-23:
The problem probably lies in the backend, since the `/upload` POST is not getting executed. The agent's upload procedure only uses the port name, thus vid and pid are totally useless if you were able to reach that point. I'll report this problem to my colleagues working on the BE, thanks!

zmoog, 2021-01-29:
Hey @zmechanic do you still have this need? If you can describe a compelling use we can put a task in the backlog. Let us know!

closed 2021-03-01 by github-actions[bot]