arduino/arduino-create-agent #157: macOS command line install

alwillis, 2017-07-03:
How do we install this on macOS via the command line? The usual `sudo installer -pkg` doesn't work, since `ArduinoCreateAgent-1.0-osx-installer.app` isn’t a package.

Doing `sudo open ArduinoCreateAgent-1.0-osx-installer.app` doesn’t seem to work either.

Suggestions?

facchinm, 2017-07-04:
I'm not an OSX expert, but since the installer is created using Installbuilder I think [this guide](https://support.bitrock.com/hc/en-us/community/posts/115002040785-MAC-OS-Command-line-argument-to-run-the-installer-in-silent-mode-) should apply (once you mount the `dmg` of course). Feel free to comment the issue if this method solves it :wink:

zmoog, 2021-01-29:
Hey @alwillis, are you still using the Create Agent? 

You can download and run the latest installer from https://github.com/arduino/arduino-create-agent/releases/

closed 2021-01-29 by zmoog