arduino/arduino-create-agent #161: debug console does not work on firefox

mastrolinux, 2017-07-07:
I just tested FF version 54.0 and the agent does not display any info in the debug console at https://localhost:8992

The form is present but there is no output.

mastrolinux, 2017-07-07:
```TypeError: msg.indexOf is not a function[Learn More]  localhost:8992:19:50
	appendLog https://localhost:8992/:19:50
	<anonymous> https://localhost:8992/:53:13
	[9]</n.prototype.emit https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:13388
	[5]</r.prototype.emit https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:8384
	[5]</r.prototype.onclose https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:8961
	[8]</e.exports/< https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:12320
	[9]</n.prototype.emit https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:13388
	[3]</n.prototype.onclose https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:6040
	[8]</e.exports/< https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:12320
	[9]</n.prototype.emit https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:13388
	[13]</</r.prototype.onClose https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:23258
	[13]</</r.prototype.setTransport/< https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:18049
	[9]</n.prototype.emit https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:13388
	[14]</n.prototype.onClose https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:1:24974
	[19]</n.prototype.onClose https://cdnjs.cloudflare.com/ajax/libs/socket.io/1.3.5/socket.io.min.js:2:3612
	[19]</n.prototype.addEventListeners/this.ws.onclose```

BallscrewBob, 2017-08-17:
Should that not be 8991 ?

hannobraun, 2018-01-29:
The debug console works for me in Firefox 58.0 on Window 10 and Linux.

@mastrolinux Can you still reproduce?

zmoog, 2021-02-02:
Works on Linux in Firefox 84.0.2. 

Closing this issue, feel free to comment or reopen.

closed 2021-02-02 by zmoog