arduino/arduino-create-agent #177: Create installer after compiling with different config.ini

baseljd, 2017-09-15:
After compiling the source, how can I make an installer for the Agent? Just changing the config.ini and copying it to another machine doesn't work. I guess I have to install some certificates and configure some files on the machine for it to work.

I am trying to change the config.ini to give access to my server. I then want to give the user the new installer that has my updated config.ini

After compiling the agent using `go build` I get a **arduino-create-agent** file.
Running it would gives an error like so **cannot open config file**. So i copied it into the source in the same directory of the config.ini and i'm getting this error 

```
INFO[0000] Your serial ports:                           
INFO[0000] 	There are no serial ports to list.          
INFO[0000] You can enter verbose mode to see all logging by starting with the -v command line switch. 
[GIN-debug] [WARNING] Running in "debug" mode. Switch to "release" mode in production.
 - using env:	export GIN_MODE=release
 - using code:	gin.SetMode(gin.ReleaseMode)

[GIN-debug] GET   /                         --> main.homeHandler (2 handlers)
Failed to write to log, short write
[GIN-debug] GET   /certificate.crt          --> main.certHandler (2 handlers)
[GIN-debug] DELETE /certificate.crt          --> main.deleteCertHandler (2 handlers)
[GIN-debug] POST  /upload                   --> main.uploadHandler (2 handlers)
[GIN-debug] GET   /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] POST  /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] WS    /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] WSS   /socket.io/               --> main.(*WsServer).ServeHTTP-fm (2 handlers)
[GIN-debug] GET   /info                     --> main.infoHandler (2 handlers)
[GIN-debug] POST  /killbrowser              --> main.killBrowserHandler (2 handlers)
[GIN-debug] POST  /pause                    --> main.pauseHandler (2 handlers)
[GIN-debug] POST  /update                   --> main.updateHandler (2 handlers)
[GIN-debug] Listening and serving HTTP on 127.0.0.1:8991
[ERR] bonjour: Failed to bind to udp6 port: listen udp6 :5353: bind: address already in use
Failed to write to log, short write
Failed to write to log, short write
panic: reflect: call of reflect.Value.Int on zero Value

goroutine 51 [running]:
reflect.Value.Int(0x0, 0x0, 0x0, 0x45f7747)
	/Users/Basel/Downloads/go/src/reflect/value.go:908 +0x141
github.com/arduino/arduino-create-agent/vendor/golang.org/x/net/ipv4.sysfd(0x48f4a60, 0xc4202d2000, 0xc4202d2000, 0x48f4a60, 0xc4202d2000)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/vendor/golang.org/x/net/ipv4/helper_unix.go:46 +0x196
github.com/arduino/arduino-create-agent/vendor/golang.org/x/net/ipv4.(*dgramOpt).sysfd(0xc4202b4100, 0x0, 0x0, 0xa)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/vendor/golang.org/x/net/ipv4/helper_unix.go:25 +0x7f
github.com/arduino/arduino-create-agent/vendor/golang.org/x/net/ipv4.(*dgramOpt).JoinGroup(0xc4202b4100, 0xc42028d708, 0x48ed8c0, 0xc42028d678, 0x0, 0xc4202b40a0)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/vendor/golang.org/x/net/ipv4/dgramopt_posix.go:109 +0x49
github.com/arduino/arduino-create-agent/vendor/github.com/oleksandr/bonjour.newClient(0x0, 0x0, 0x0, 0x0)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/vendor/github.com/oleksandr/bonjour/client.go:115 +0x632
github.com/arduino/arduino-create-agent/vendor/github.com/oleksandr/bonjour.NewResolver(0x0, 0x0, 0x0, 0x0)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/vendor/github.com/oleksandr/bonjour/client.go:23 +0x2f
main.getPorts(0x0, 0x0, 0x0, 0x0, 0x0)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/discovery.go:111 +0x3c
main.GetNetworkList(0x0, 0x0, 0x0, 0x0, 0x0)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/discovery.go:49 +0x34
main.GetList(0x1, 0x0, 0x0, 0x0, 0x0, 0x0)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/seriallist.go:28 +0x8cb
main.spListDual(0x1)
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/serial.go:147 +0x42
main.discoverLoop.func2()
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/serial.go:138 +0x26
created by main.discoverLoop
	/Users/Basel/Downloads/go/src/github.com/arduino/arduino-create-agent/serial.go:136 +0x103
```

matteosuppo, 2017-09-15:
You're getting the panic because you are compiling with golang 1.9, and there's an issue with that version. With 1.8 you'll have no trouble.

We cannot share the way to create an installer since it involves paid licenses.

You can however generate the certificates with the commands here: https://github.com/arduino/arduino-create-agent/blob/devel/certificates.go

zmoog, 2021-01-15:
We're currently using Go 1.14 to build the project, so I think this issue no longer applies.

Fell free to reopen the issue if this error still occurs in you case.

closed 2021-01-15 by zmoog