arduino/arduino-create-agent #179: Please reselect board

BallscrewBob, 2017-09-17:
Issue is affecting multiple browsers and OS.
Currently at least 3 or more threads in the forums.
 
[https://forum.arduino.cc/index.php?topic=500456.msg3414809#new](url)

Affects both real and clone boards but oddly not all of them.
Don't have much more to give you other than the forums posts themselves at this time.
I currently only have a REAL UNO affected.

per1234, 2017-09-17:
Where did you get this "Please reselect board"?. That isn't mentioned anywhere in either of the threads. The error is:
```
The tool is already present on the system
```
It occurs when uploading. Compilation is not affected.

Here's another thread about this:
https://forum.arduino.cc/index.php?topic=500428
And the fixed link to the thread mentioned above:
https://forum.arduino.cc/index.php?topic=500456

BallscrewBob, 2017-09-18:
My bad and Thanks Pert for adding the correct links.

smellai, 2017-09-18:
Everything should be ok now, seems that the last release of the agent was breaking the client
Sorry

closed 2017-09-18 by smellai