arduino/arduino-create-agent #183: not enough arguments in call to systray.Run when building

shawt, 2017-09-26:
I'm new to this, so perhaps I'm doing something wrong, but when I follow the instructions to build the project, after I run go get, I get the following error:

trayicon.go:46: not enough arguments in call to systray.Run

Looks like the call on line 46 of trayicon.go is only including one argument, but the systray.Run function in the getlantern project seems to want two arguments.

Any help would be appreciated.

facchinm, 2017-09-26:
Hi @shawt ,
it looks like your go toolchain is not considering the `vendor` folder. We vendored it to avoid this king of problems, with mutable dependencies. The file that should be used is [here](https://github.com/arduino/arduino-create-agent/blob/devel/vendor/github.com/getlantern/systray/systray.go#L49)

The information on "[how to build](https://github.com/arduino/arduino-create-agent#development)" in the README are probably a bit outdated, so I'd suggest you to:
- delete your current cloned repo
- create a folder ("arduino-create-agent" for example)
- `cd` in that folder
- run `export GOPATH=$PWD` 
- run `go get github.com/arduino/arduino-create-agent`
- run `go build github.com/arduino/arduino-create-agent`

This should avoid any problem with updated dependencies. Hope it helps!

shawt, 2017-09-26:
Yes. That did the trick. Thank you!!

closed 2017-09-26 by shawt