arduino/arduino-create-agent #187: Programmer not responding when flashing to Leonardo

shawt, 2017-10-18:
Implementing this tool with my own site. Things work great when flashing to an Uno or a Lilypad using the arduino programmer.

However, when I try to flash a Leonardo, I get a programmer status:busy and programmer not responding error.

I am sending the following object:

`{
"board":"arduino:avr:leonardo",
"port":"/dev/cu.usbmodem1411",
"commandline":"\"{runtime.tools.avrdude.path}/bin/avrdude\" \"-C{runtime.tools.avrdude.path}/etc/avrdude.conf\" -patmega32u4 -cavr109 -P{serial.port} -b57600 -D \"-Uflash:w:{build.path}/{build.project_name}.hex:i\"",

"signature":"66a7afc961598...",
"extra":{"auth":
 {"username":null,
"password":null},
"use_1200pbs_touch":true,
"wait_for_upload_port":true,
"verbose":true,
"params_verbose":"-v",
"params_quiet":"-q -q",
"network":false},
"hex":"OjEwMDA...",
"filename":"toFlash.hex"}`

When I use the Agent with the Arduino-Create web editor, it seems to work and the JSON object produced there seems to match this one. I can also flash with the same command line from the Arduino IDE.

Can anyone give any insight into what else I might need to do to get my web site working with a leonardo?

Again, Uno and Lillypad work fine.

Thanks for any help.

facchinm, 2017-10-18:
Hi @shawt ,
are you getting any error on the debug console? A silent output usually means that the signature is wrong, so the upload won't even start. 

The fact that it only happens with the  Leonardo could also be due to the 1200bps touch on the serial port. Try forcing bootloader mode by double pressing the reset button while the upload is ongoing to check if it does something.

shawt, 2017-10-18:
Hi @facchinm ,
Thanks for the reply. I'm pretty sure the sig is fine, since I can flash to other boards.  I get the following messages from the agent which I echo to my console screen.

`time="2017-10-18T08:27:18-04:00" level=info msg="Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding" 

iForgeLibraries.js:1379 {"Msg":"Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding","ProgrammerStatus":"Busy"}
iForgeLibraries.js:1379 time="2017-10-18T08:27:18-04:00" level=info msg= 

iForgeLibraries.js:1379 {"Msg":"","ProgrammerStatus":"Busy"}

time="2017-10-18T08:27:44-04:00" level=info msg="Found programmer: Id = \"\"; type = \x00" 

iForgeLibraries.js:1379 {"Msg":"Found programmer: Id = \"\"; type = \u0000","ProgrammerStatus":"Busy"}
iForgeLibraries.js:1379 time="2017-10-18T08:27:44-04:00" level=info msg="    Software Version = \x00.\xe1; Hardware Version = \x00.\x00" 

iForgeLibraries.js:1379 {"Msg":"    Software Version = \u0000.\ufffd; Hardware Version = \u0000.\u0000","ProgrammerStatus":"Busy"}`

The 1200bps touch was a shot in the dark. This seems to be what the arduino web editor sends, and I was trying to match it exactly.  Hitting reset during upload doesn't seem to have any effect.

Thanks again for any advice.

Trevor

facchinm, 2017-10-18:
Usually `programmer is not responding` means that it failed to reboot the board in bootloader mode, and it still tries to upload on the "sketch" serial port where the commands are not getting recognized. The strange thing is that Create works, otherwise I would advise you to try changing USB port...

shawt, 2017-10-18:
Thanks,
I edited my previous reply to add the output I get when I try to force bootloader mode. Not sure if that offers anything helpful.

shawt, 2017-10-18:
Tried different serial ports, different cables, different computers and different Leonardo boards.  I'm stumped.

shawt, 2017-10-18:
I just tried from the command line and I'm getting the same errors. Looks like something I'm doing wrong with avrdude, maybe. I will try to post to a forum on that and maybe stackoverflow. I will update as soon as I can get it working from the command line. I suspect that will take care of it.

shawt, 2017-10-18:
Okay. I'm pretty sure I figured it out. In case anyone else runs into same...

As per this page: https://nicholaskell.wordpress.com/tag/leonardo/ I need to trigger bootloader mode by making a serial connection with 1200bps. The reason I wasn't able to make this work by tapping the reset button is that the leonardo often grabs a new serial port name when it resets, so I would enter bootloader successfully, but when I went to flash, the port I was using was no longer valid. In my web page, I now connect to serial @1200bps. When I disconnect, the board enters bootloader mode. I then scan my ports to make sure I have a valid port and upload using that valid port.  Still lots of glitchiness to iron out, but I got a proof of concept working.

Thanks for all your help.

closed 2017-10-18 by shawt