arduino/arduino-create-agent #189: Root certificate required?

vatgh, 2017-11-15:
When installing the Windows version I got a warning that it wants to install a root certificate. I declined. Is this really required? The warning is certainly a bit worrisome in particular because it is a root certificate and might not be limited to the use in the context of Arduino.

facchinm, 2017-11-15:
Hi @vatgh ,
I know it can be scary but it's the only way to allow a piece of Create webpage (which is all behind https) to embed a localhost secure websocket (talking with the agent).
We remove the root certificate as soon as we generate the derived keys used by the agent, so an attacker cannot sign other stuff with it.

umbynos, 2021-02-05:
Hi, we provide different flavors of the agent for different browsers: the version for chrome/firefox asks if you use also other browsers: If you click yes, it installs the https certs.
If you install the version for edge: it automatically installs them.

closed 2021-02-05 by umbynos