arduino/arduino-create-agent #190: Freeze / uninstall

Cross22, 2017-11-17:
I installed the arudino-create-agent on OSX 10.12.6 and now plugging in an Arduino nano causes my system to immediately reboot.
I ran the uninstaller under ~/Applications and those files are gone, however, that did not fix the problem as the computer still reboots. So how do I completely undo the changes that this POS made to my system?

Cross22, 2017-11-18:
The root cause appears to be an outdated driver: https://github.com/adrianmihalko/ch340g-ch34g-ch34x-mac-os-x-driver
It's not clear though why the crash didn't happen until after I installed create-agent.

facchinm, 2017-11-20:
None of the official Arduino boards ship with CH3* serial-to-usb converter, so we don't install that driver. In fact, on OSX we don't install any extra drivers since everything is already correctly handled by the OS so the timing must be a coincidence.

closed 2017-11-20 by facchinm