arduino/arduino-create-agent #191: Add ARM build platform (v6/v7) to the supported platforms

marty-buly, 2017-11-27:
Once provided, one can program Arduino from e.g. Raspberry Pi 3 with just browser. This way RPi users will get most recent version of Arduino "IDE" by default. (Latest supported Arduino IDE is still version 1.0.1 or so...)

facchinm, 2017-11-28:
Hi @marty-buly ,
latest supported IDE on RPi is 1.8.5 but the Debian packagers are not providing it in their repos.
About the agent, the main problem in providing an ARM build is that Installbulder (that we use to create the installer) is unable to produce installers for LinuxARM platforms.
If you wish, I can provide a CLI-only build with almost no effort, but you would need to install the certificates (and do everything the installer does) manually.

marty-buly, 2017-11-28:
Hi @facchinm,
I was not aware of the direct download for ARM of Arduino IDE, thxn for pointing out (and apologize for spamming when it is clearly my fault not to discover :)). It will be sufficient for me now. Maybe in the future I can try manual approach you mentioned.

closed 2017-11-28 by marty-buly