arduino/arduino-create-agent #192: Uninstaller on Mac didn't remove added root certificates

mblackshaw, 2017-11-29:
Just tried uninstalled Create and it left the "Arduino" root certificates in my keychain. :(

facchinm, 2017-12-01:
Hi @mblackshaw ,
thanks for reporting. Did you uninstall the agent using the uninstaller or moving the app into Trash a-la-osx ?

mblackshaw, 2017-12-01:
I used the uninstaller – and then noticed the certificates were still in the keychain afterwards.

facchinm, 2017-12-04:
Thanks, I'm investigating it today; thank you for reporting!

umbynos, 2021-02-05:
Just tested with [1.2.0](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.0) on macOS 11.2 and I can confirm that the certificate get correctly removed during uninstall. Fixed in https://github.com/bcmi-labs/arduino-create-agent-installer/pull/2

closed 2021-02-05 by umbynos