arduino/arduino-create-agent #193: Terminal stays open after build.

Vancer, 2017-12-18:
So after I have configured the _config.ini_ file and made some changes to the code I have decided to build. Everything is going well and the agent is communicating with my server, Cool!

There is something bugging me tho, whenever I run the agent, the debug terminal pops up and whenever i close it the agent shuts off. I've tried playing with the verbosity flags in the main.go file and in the config.ini and nothing worked. I'm trying to compile the agent and make it run without the terminal showing. 

Note that I am on Windows and I'm using Git Bash, Ive tried exporting the GIN path parameter with a value of release but nothing worked, i tried doing it using the code too but no clue...

Any help would be appreciated, Thanks Arduino Team!

facchinm, 2017-12-18:
Hi @Vancer, to compile the agent without a terminal popping up (it's a Windows specific issue) you should compile it with 
```
go build -ldflags "-H=windowsgui" 
```
Feel free to close the issue if it works as intended :wink: 
Cheers!

Vancer, 2017-12-21:
Thanks, Works like a charm!

closed 2017-12-21 by Vancer