arduino/arduino-create-agent #212: MAC OS - Error deleting service ArduinoCreateAgent

smellai, 2018-02-12:
http://forum.arduino.cc/index.php?topic=528108.new#new

raej3158, 2019-09-12:
I had  a problem regarding the deletion of the arduino create agent, however upon fixing and trying to delete, I found an error [url=https://www.mactechnicalsupportnumbers.com/blog/get-easy-fixes-for-mac-error-code-43/]Mac Error Code 43[/url], which you can do is clear the time machine virtual disk and reboot the system, accordingly if then also you can delete the service you have to force shut off the program .

crislee01, 2019-12-18:
This error message appears while trying to install iTunes on their PC, and one way to fix this problem is to rename one of your folders. According to users, com.Apple.Outlook.client.resources folder can cause this error to appear, therefore in order to fix this problem you need to find this folder and rename it to com.Apple.Outlook.client.resources_OLD for example.

After renaming the folder, try to perform the installation again and check if your problem is resolved.
i faced almost the same problem but its comes with [iTunes Error 9006](https://iphonesupportnumber.com/blog/itunes-error-9006/)

per1234, 2019-12-18:
I reported @crislee01's spam comment to GitHub. @raej3158 also looks very suspicious to me.

KateDunk, 2020-01-16:
Hi,
I am also facing an issue of [Mac Error Code 43](url) in my newly purchased Macbook Pro. Is there a permanent resolution to this, as I am unable to delete files permanently and also I cannot move it to trash.

umbynos, 2021-02-05:
Hi, could you please try the latest version? [1.2.0](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.0)

closed 2021-03-01 by github-actions[bot]