arduino/arduino-create-agent #215: POST request - Upload hex file to arduino from Angular 5

juanfjk, 2018-03-20:
Hello

I used postman to send the request an I got 202 status code and the program uploaded to my Arduino successfully. I'm coding an Angular 5 app, but when I tried to make the request from Angular I get a 200 status code and nothing happens in my board. I compared Postman request with Angular request and they don't have differences. I don' understand what happened with Angular Request. Can you give me an idea to solve this issue, please? I'll appreciate your help.

**This is Postman request [202]**
````
POST /upload HTTP/1.1
Host: localhost:8991
Content-Type: text/plain; charset=utf-8
Cache-Control: no-cache

{
    "board": "arduino:avr:uno",
    "port": "/dev/ttyUSB0",
    "commandline": "\"{runtime.tools.avrdude.path}/bin/avrdude\" \"-C{runtime.tools.avrdude.path}/etc/avrdude.conf\" {upload.verbose}  -patmega328p -carduino \"-P{serial.port}\" -b115200 -D \"-Uflash:w:{build.path}/{build.project_name}.hex:i\"",
    "signature": "6596474c3c77cb52f4441c1bdc219abd0a521de0f852652105f1af1f650df0bcdf2a73739928f1863f134e5b4c763e77e351a5a503eb2e128ce8d99e4472ab2356d5357506cb219d15888ebb50025f7711c9124f041c18455818bace18a710f976e188aa45cea9e8187083c493e936323e4128573d2d22b4859df0ef4fe135f8f5ba35ef57bc06a9a15ac6dfddffa2063813b021e94ea64c823bcb1782a2c4ae3e091570826ec7324066896d0104262fc4063a56ae33de738e5583360dd7ec0407551f59f7e9e9a65d93a25ca516a73726c622a78d8a6009dccf23753b7c072b5249aebfd4696d3227fa0571b1bf430a3318757850930095e5dd585691493a9b",
    "hex": "OjEwMDAwMDAwMEM5NDVDMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDBDQQ0KOjEwMDAxMDAwMEM5NDZFMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDBBOA0KOjEwMDAyMDAwMEM5NDZFMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDA5OA0KOjEwMDAzMDAwMEM5NDZFMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDA4OA0KOjEwMDA0MDAwMEM5NDFEMDEwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDBDOA0KOjEwMDA1MDAwMEM5NDZFMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDA2OA0KOjEwMDA2MDAwMEM5NDZFMDAwQzk0NkUwMDAwMDAwMDA3MDAwMjAxMDA2QQ0KOjEwMDA3MDAwMDAwMzA0MDYwMDAwMDAwMDAwMDAwMDAwMDEwMjA0MDg2NA0KOjEwMDA4MDAwMTAyMDQwODAwMTAyMDQwODEwMjAwMTAyMDQwODEwMjAwMg0KOjEwMDA5MDAwMDQwNDA0MDQwNDA0MDQwNDAyMDIwMjAyMDIwMjAzMDMyRQ0KOjEwMDBBMDAwMDMwMzAzMDMwMDAwMDAwMDI1MDAyODAwMkIwMDAwMDBDQw0KOjEwMDBCMDAwMDAwMDI0MDAyNzAwMkEwMDExMjQxRkJFQ0ZFRkQ4RTA0Mw0KOjEwMDBDMDAwREVCRkNEQkYxMUUwQTBFMEIxRTAwMUMwMUQ5MkE5MzBCQw0KOjEwMDBEMDAwQjEwN0UxRjcwRTk0MTAwMTBDOTRBMjAxMEM5NDAwMDBGQQ0KOjEwMDBFMDAwNjFFMDgyRTAwQzk0QTEwMDYxRTA4MkUwMEM5NERBMDAwRg0KOjEwMDBGMDAwODMzMDgxRjAyOEY0ODEzMDk5RjA4MjMwQTFGMDA4OTVBNg0KOjEwMDEwMDAwODYzMEE5RjA4NzMwQjlGMDg0MzBEMUY0ODA5MTgwMDAzNg0KOjEwMDExMDAwOEY3RDAzQzA4MDkxODAwMDhGNzc4MDkzODAwMDA4OTU0OQ0KOjEwMDEyMDAwODRCNThGNzcwMkMwODRCNThGN0Q4NEJEMDg5NTgwOTE5QQ0KOjEwMDEzMDAwQjAwMDhGNzcwM0MwODA5MUIwMDA4RjdEODA5M0IwMDBCNg0KOjEwMDE0MDAwMDg5NUNGOTNERjkzOTBFMEZDMDFFNDU4RkY0RjI0OTE5Mg0KOjEwMDE1MDAwRkMwMUUwNTdGRjRGODQ5MTg4MjM0OUYxOTBFMDg4MEYxQw0KOjEwMDE2MDAwOTkxRkZDMDFFMjU1RkY0RkE1OTFCNDkxOEM1NTlGNEYwQg0KOjEwMDE3MDAwRkMwMUM1OTFENDkxOUZCNzYxMTEwOEMwRjg5NDhDOTE4RQ0KOjEwMDE4MDAwMjA5NTgyMjM4QzkzODg4MTgyMjMwQUMwNjIzMDUxRjRBNw0KOjEwMDE5MDAwRjg5NDhDOTEzMjJGMzA5NTgzMjM4QzkzODg4MTgyMkIxNQ0KOjEwMDFBMDAwODg4MzA0QzBGODk0OEM5MTgyMkI4QzkzOUZCRkRGOTEzRA0KOjEwMDFCMDAwQ0Y5MTA4OTUwRjkzMUY5M0NGOTNERjkzMUY5MkNEQjdFNQ0KOjEwMDFDMDAwREVCNzI4MkYzMEUwRjkwMUU4NTlGRjRGODQ5MUY5MDE5Qg0KOjEwMDFEMDAwRTQ1OEZGNEYxNDkxRjkwMUUwNTdGRjRGMDQ5MTAwMjNCOQ0KOjEwMDFFMDAwQzlGMDg4MjMyMUYwNjk4MzBFOTQ3ODAwNjk4MUUwMkY5Qg0KOjEwMDFGMDAwRjBFMEVFMEZGRjFGRUM1NUZGNEZBNTkxQjQ5MTlGQjdCNA0KOjEwMDIwMDAwRjg5NDhDOTE2MTExMDNDMDEwOTU4MTIzMDFDMDgxMkI1QQ0KOjEwMDIxMDAwOEM5MzlGQkYwRjkwREY5MUNGOTExRjkxMEY5MTA4OTUwNQ0KOjEwMDIyMDAwMEU5NDY3MDEwRTk0NzAwMEMwRTBEMEUwMEU5NDc0MDA0Qw0KOjEwMDIzMDAwMjA5N0UxRjMwRTk0MDAwMEY5Q0YxRjkyMEY5MjBGQjZCMg0KOjEwMDI0MDAwMEY5MjExMjQyRjkzM0Y5MzhGOTM5RjkzQUY5M0JGOTM1Qw0KOjEwMDI1MDAwODA5MTAxMDE5MDkxMDIwMUEwOTEwMzAxQjA5MTA0MDFFQw0KOjEwMDI2MDAwMzA5MTAwMDEyM0UwMjMwRjJEMzcyMEY0MDE5NkExMURDQQ0KOjEwMDI3MDAwQjExRDA1QzAyNkU4MjMwRjAyOTZBMTFEQjExRDIwOTNENA0KOjEwMDI4MDAwMDAwMTgwOTMwMTAxOTA5MzAyMDFBMDkzMDMwMUIwOTNCOA0KOjEwMDI5MDAwMDQwMTgwOTEwNTAxOTA5MTA2MDFBMDkxMDcwMUIwOTFBMA0KOjEwMDJBMDAwMDgwMTAxOTZBMTFEQjExRDgwOTMwNTAxOTA5MzA2MDFERg0KOjEwMDJCMDAwQTA5MzA3MDFCMDkzMDgwMUJGOTFBRjkxOUY5MThGOTFENw0KOjEwMDJDMDAwM0Y5MTJGOTEwRjkwMEZCRTBGOTAxRjkwMTg5NTc4OTQyQg0KOjEwMDJEMDAwODRCNTgyNjA4NEJEODRCNTgxNjA4NEJEODVCNTgyNjA0Qg0KOjEwMDJFMDAwODVCRDg1QjU4MTYwODVCREVFRTZGMEUwODA4MTgxNjBFOQ0KOjEwMDJGMDAwODA4M0UxRThGMEUwMTA4MjgwODE4MjYwODA4MzgwODFFOQ0KOjEwMDMwMDAwODE2MDgwODNFMEU4RjBFMDgwODE4MTYwODA4M0UxRUJDMA0KOjEwMDMxMDAwRjBFMDgwODE4NDYwODA4M0UwRUJGMEUwODA4MTgxNjBBOA0KOjEwMDMyMDAwODA4M0VBRTdGMEUwODA4MTg0NjA4MDgzODA4MTgyNjA1RQ0KOjEwMDMzMDAwODA4MzgwODE4MTYwODA4MzgwODE4MDY4ODA4MzEwOTI0Nw0KOjBFMDM0MDAwQzEwMDA4OTVGODk0MEM5NEE1MDFGODk0RkZDRjI1DQo6MDAwMDAwMDFGRg0K",
    "filename": "sketch_mar12a.hex",
    "extra": {
        "auth": {},
        "wait_for_upload_port": false,
        "use_1200bps_touch": false,
        "network": false
    },
    "extrafiles": []
}
````

**This is Angular 5 request [200]**
[image]

matteosuppo, 2018-03-22:
You should add your origin in the `config.ini` Browsers are more picky than postman when external requests are concerned. See this for more informations: https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS

juanfjk, 2018-03-23:
@matteosuppo thanks for your anwser. It solved with your instruction.

closed 2018-03-23 by matteosuppo

agentpx, 2021-04-24:
> 
> 
> Hello
> 
> I used postman to send the request an I got 202 status code and the program uploaded to my Arduino successfully. I'm coding an Angular 5 app, but when I tried to make the request from Angular I get a 200 status code and nothing happens in my board. I compared Postman request with Angular request and they don't have differences. I don' understand what happened with Angular Request. In angular, I'm working in development enviroment so my origin is localhost:4200. Can you give me an idea to solve this issue, please? I'll appreciate your help.
> 
> **This is Postman request [202]**
> 
> ```
> POST /upload HTTP/1.1
> Host: localhost:8991
> Content-Type: text/plain; charset=utf-8
> Cache-Control: no-cache
> 
> {
>     "board": "arduino:avr:uno",
>     "port": "/dev/ttyUSB0",
>     "commandline": "\"{runtime.tools.avrdude.path}/bin/avrdude\" \"-C{runtime.tools.avrdude.path}/etc/avrdude.conf\" {upload.verbose}  -patmega328p -carduino \"-P{serial.port}\" -b115200 -D \"-Uflash:w:{build.path}/{build.project_name}.hex:i\"",
>     "signature": "6596474c3c77cb52f4441c1bdc219abd0a521de0f852652105f1af1f650df0bcdf2a73739928f1863f134e5b4c763e77e351a5a503eb2e128ce8d99e4472ab2356d5357506cb219d15888ebb50025f7711c9124f041c18455818bace18a710f976e188aa45cea9e8187083c493e936323e4128573d2d22b4859df0ef4fe135f8f5ba35ef57bc06a9a15ac6dfddffa2063813b021e94ea64c823bcb1782a2c4ae3e091570826ec7324066896d0104262fc4063a56ae33de738e5583360dd7ec0407551f59f7e9e9a65d93a25ca516a73726c622a78d8a6009dccf23753b7c072b5249aebfd4696d3227fa0571b1bf430a3318757850930095e5dd585691493a9b",
>     "hex": "OjEwMDAwMDAwMEM5NDVDMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDBDQQ0KOjEwMDAxMDAwMEM5NDZFMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDBBOA0KOjEwMDAyMDAwMEM5NDZFMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDA5OA0KOjEwMDAzMDAwMEM5NDZFMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDA4OA0KOjEwMDA0MDAwMEM5NDFEMDEwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDBDOA0KOjEwMDA1MDAwMEM5NDZFMDAwQzk0NkUwMDBDOTQ2RTAwMEM5NDZFMDA2OA0KOjEwMDA2MDAwMEM5NDZFMDAwQzk0NkUwMDAwMDAwMDA3MDAwMjAxMDA2QQ0KOjEwMDA3MDAwMDAwMzA0MDYwMDAwMDAwMDAwMDAwMDAwMDEwMjA0MDg2NA0KOjEwMDA4MDAwMTAyMDQwODAwMTAyMDQwODEwMjAwMTAyMDQwODEwMjAwMg0KOjEwMDA5MDAwMDQwNDA0MDQwNDA0MDQwNDAyMDIwMjAyMDIwMjAzMDMyRQ0KOjEwMDBBMDAwMDMwMzAzMDMwMDAwMDAwMDI1MDAyODAwMkIwMDAwMDBDQw0KOjEwMDBCMDAwMDAwMDI0MDAyNzAwMkEwMDExMjQxRkJFQ0ZFRkQ4RTA0Mw0KOjEwMDBDMDAwREVCRkNEQkYxMUUwQTBFMEIxRTAwMUMwMUQ5MkE5MzBCQw0KOjEwMDBEMDAwQjEwN0UxRjcwRTk0MTAwMTBDOTRBMjAxMEM5NDAwMDBGQQ0KOjEwMDBFMDAwNjFFMDgyRTAwQzk0QTEwMDYxRTA4MkUwMEM5NERBMDAwRg0KOjEwMDBGMDAwODMzMDgxRjAyOEY0ODEzMDk5RjA4MjMwQTFGMDA4OTVBNg0KOjEwMDEwMDAwODYzMEE5RjA4NzMwQjlGMDg0MzBEMUY0ODA5MTgwMDAzNg0KOjEwMDExMDAwOEY3RDAzQzA4MDkxODAwMDhGNzc4MDkzODAwMDA4OTU0OQ0KOjEwMDEyMDAwODRCNThGNzcwMkMwODRCNThGN0Q4NEJEMDg5NTgwOTE5QQ0KOjEwMDEzMDAwQjAwMDhGNzcwM0MwODA5MUIwMDA4RjdEODA5M0IwMDBCNg0KOjEwMDE0MDAwMDg5NUNGOTNERjkzOTBFMEZDMDFFNDU4RkY0RjI0OTE5Mg0KOjEwMDE1MDAwRkMwMUUwNTdGRjRGODQ5MTg4MjM0OUYxOTBFMDg4MEYxQw0KOjEwMDE2MDAwOTkxRkZDMDFFMjU1RkY0RkE1OTFCNDkxOEM1NTlGNEYwQg0KOjEwMDE3MDAwRkMwMUM1OTFENDkxOUZCNzYxMTEwOEMwRjg5NDhDOTE4RQ0KOjEwMDE4MDAwMjA5NTgyMjM4QzkzODg4MTgyMjMwQUMwNjIzMDUxRjRBNw0KOjEwMDE5MDAwRjg5NDhDOTEzMjJGMzA5NTgzMjM4QzkzODg4MTgyMkIxNQ0KOjEwMDFBMDAwODg4MzA0QzBGODk0OEM5MTgyMkI4QzkzOUZCRkRGOTEzRA0KOjEwMDFCMDAwQ0Y5MTA4OTUwRjkzMUY5M0NGOTNERjkzMUY5MkNEQjdFNQ0KOjEwMDFDMDAwREVCNzI4MkYzMEUwRjkwMUU4NTlGRjRGODQ5MUY5MDE5Qg0KOjEwMDFEMDAwRTQ1OEZGNEYxNDkxRjkwMUUwNTdGRjRGMDQ5MTAwMjNCOQ0KOjEwMDFFMDAwQzlGMDg4MjMyMUYwNjk4MzBFOTQ3ODAwNjk4MUUwMkY5Qg0KOjEwMDFGMDAwRjBFMEVFMEZGRjFGRUM1NUZGNEZBNTkxQjQ5MTlGQjdCNA0KOjEwMDIwMDAwRjg5NDhDOTE2MTExMDNDMDEwOTU4MTIzMDFDMDgxMkI1QQ0KOjEwMDIxMDAwOEM5MzlGQkYwRjkwREY5MUNGOTExRjkxMEY5MTA4OTUwNQ0KOjEwMDIyMDAwMEU5NDY3MDEwRTk0NzAwMEMwRTBEMEUwMEU5NDc0MDA0Qw0KOjEwMDIzMDAwMjA5N0UxRjMwRTk0MDAwMEY5Q0YxRjkyMEY5MjBGQjZCMg0KOjEwMDI0MDAwMEY5MjExMjQyRjkzM0Y5MzhGOTM5RjkzQUY5M0JGOTM1Qw0KOjEwMDI1MDAwODA5MTAxMDE5MDkxMDIwMUEwOTEwMzAxQjA5MTA0MDFFQw0KOjEwMDI2MDAwMzA5MTAwMDEyM0UwMjMwRjJEMzcyMEY0MDE5NkExMURDQQ0KOjEwMDI3MDAwQjExRDA1QzAyNkU4MjMwRjAyOTZBMTFEQjExRDIwOTNENA0KOjEwMDI4MDAwMDAwMTgwOTMwMTAxOTA5MzAyMDFBMDkzMDMwMUIwOTNCOA0KOjEwMDI5MDAwMDQwMTgwOTEwNTAxOTA5MTA2MDFBMDkxMDcwMUIwOTFBMA0KOjEwMDJBMDAwMDgwMTAxOTZBMTFEQjExRDgwOTMwNTAxOTA5MzA2MDFERg0KOjEwMDJCMDAwQTA5MzA3MDFCMDkzMDgwMUJGOTFBRjkxOUY5MThGOTFENw0KOjEwMDJDMDAwM0Y5MTJGOTEwRjkwMEZCRTBGOTAxRjkwMTg5NTc4OTQyQg0KOjEwMDJEMDAwODRCNTgyNjA4NEJEODRCNTgxNjA4NEJEODVCNTgyNjA0Qg0KOjEwMDJFMDAwODVCRDg1QjU4MTYwODVCREVFRTZGMEUwODA4MTgxNjBFOQ0KOjEwMDJGMDAwODA4M0UxRThGMEUwMTA4MjgwODE4MjYwODA4MzgwODFFOQ0KOjEwMDMwMDAwODE2MDgwODNFMEU4RjBFMDgwODE4MTYwODA4M0UxRUJDMA0KOjEwMDMxMDAwRjBFMDgwODE4NDYwODA4M0UwRUJGMEUwODA4MTgxNjBBOA0KOjEwMDMyMDAwODA4M0VBRTdGMEUwODA4MTg0NjA4MDgzODA4MTgyNjA1RQ0KOjEwMDMzMDAwODA4MzgwODE4MTYwODA4MzgwODE4MDY4ODA4MzEwOTI0Nw0KOjBFMDM0MDAwQzEwMDA4OTVGODk0MEM5NEE1MDFGODk0RkZDRjI1DQo6MDAwMDAwMDFGRg0K",
>     "filename": "sketch_mar12a.hex",
>     "extra": {
>         "auth": {},
>         "wait_for_upload_port": false,
>         "use_1200bps_touch": false,
>         "network": false
>     },
>     "extrafiles": []
> }
> ```
> 
> **This is Angular 5 request [200]**
> [image]
> 
> **It's my Angular5 POST request implementation**
> 
> ```
>   downloadCode(): Observable<any> { 
> 
>     let url = 'http://localhost:8991/upload';
>     let headers = new Headers({ 'Content-Type' : 'text/plain; charset=utf-8' }); 
>     let options = new RequestOptions({ headers: headers });
> 
>     let temp = {
>       "board":"arduino:avr:uno",
>       "port":"/dev/ttyUSB0",
[... cut ...] TI0Nw0KOjBFMDM0MDAwQzEwMDA4OTVGODk0MEM5NEE1MDFGODk0RkZDRjI1DQo6MDAwMDAwMDFGRg0K",
>       "filename":"sketch_mar12a.hex",
>       "extra":{
>         "auth":{},
>         "wait_for_upload_port":false,
>         "use_1200bps_touch":false,
>         "network":false
>       },
>       "extrafiles":[]
>     };
> 
>     return this.http.post(url, temp, options)
>   }
> ```
> 
> PD. When I call this service I get in my web developer console this message
> 
> [image]

How did you sign your commandline?