arduino/arduino-create-agent #218: Lock serial monitor close and open / add checks

smellai, 2018-04-10:
Possible bug when sending **open** when **close** in progress and not yet completed.
Possible similar cases.

I tried to add checks in the frontend but it doesn't seem to fix the issues. In some cases compilation/upload stays pending or ends with 'touch error'

smellai, 2018-04-10:
The agent could better control serial monitor connections and disconections in general, it could improve this too for example: https://github.com/bcmi-labs/webide/issues/2933

umbynos, 2021-02-05:
To solve this issue we could better handle the open/close.
Something could be offloaded from the frontend: When the WebSocket is closed the agent could close all the ports