arduino/arduino-create-agent #219: error: go get github.com/arduino/arduino-create-agent

focalhot, 2018-04-11:
```
go build github.com/arduino/arduino-create-agent/vendor/go.bug.st/serial.v1/enumerator: invalid flag in #cgo LDFLAGS: -fconstant-cfstrings
```

focalhot, 2018-04-11:
@facchinm 
```
go version go1.10.1 darwin/amd64
macOS 10.13.3
```

facchinm, 2018-04-11:
Can you try exporting
```
export MACOSX_DEPLOYMENT_TARGET=10.9
export CGO_ENABLED=1
```
in the same shell before building?
It looks like that flag is supported by latest clang so maybe it's a go toolchain bug.
We are currently building with 1.8.3 (using osxcross for OSX build)

focalhot, 2018-04-11:
@facchinm It still does not work well.Same error.

glowflame, 2018-05-15:
Any updates here? I'm having the same error: 

go version go1.10.2 darwin/amd64
MacOS 10.13.2

facchinm, 2018-11-28:
I think latest master branch needs golang 1.10+ . @rsora can you confirm?

rsora, 2018-11-28:
Confirmed, with dependencies managed with `dep` tool

My apologies for the lack of details in the README, I'll update the instructions asap!

murilopolese, 2018-12-03:
Upgrading to `1.10.5` did passed the initial problem but still getting:

```
$ go get github.com/arduino/arduino-create-agent
# pkg-config --cflags gtk+-3.0 appindicator3-0.1
Package gtk+-3.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gtk+-3.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gtk+-3.0' found
Package appindicator3-0.1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `appindicator3-0.1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'appindicator3-0.1' found
pkg-config: exit status 1

$ sudo apt install libappindicator3-1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libappindicator3-1 is already the newest version (12.10.1+18.04.20180322.1-0ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

$ sudo apt install libgtk-3-0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgtk-3-0 is already the newest version (3.22.30-1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

I, and certainly more people on the community, would really appreciate if the build instructions were a bit richer... I'm willing to contribute writing documentation, debugging those issues and in any other way you think it's possible. 

But I'm new to Go and since I am not able to "go get" the project, I'm a bit lost on where to start haha What would you suggest me to do? :)

facchinm, 2018-12-04:
Hi @murilopolese , sorry for the lack of documentation. The problem is that every OS has different prerequisites due to the tray icon being OS specific. 
Linux in particular requires libappindicator which has two (three) different names :slightly_smiling_face: 
The problem you are encountering can be solved by installing `-dev` packages, which provide all the needed headers to compile stuff against the library.

murilopolese, 2018-12-07:
Thanks big time! I managed to build! :heart: I made a small PR with updating the docs about the go version and `dev` packages as a way to say thanks!

All the best!