arduino/arduino-create-agent #224: EXEC error

BallscrewBob, 2018-05-04:
I will reference the post on the forum [ https://forum.arduino.cc/index.php?topic=543221.0;topicseen  ](url)
It does not appear to be cache related as I had cleaned up twice in seeing this error.
`
./opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -build-path /tmp/672686874/build -hardware opt/arduino-builder/hardware -hardware ./opt/cores -tools opt/arduino-builder/tools -tools ./opt/tools -built-in-libraries opt/libraries/latest -libraries /tmp/672686874/pinned -libraries /tmp/672686874/custom -fqbn arduino:avr:yun -build-cache /tmp -logger humantags -verbose=true /tmp/672686874/exploring_slowLinks

Using board 'yun' from platform in folder: /home/ubuntu/opt/cores/arduino/avr

Using core 'arduino' from platform in folder: /home/ubuntu/opt/cores/arduino/avr

Detecting libraries used...

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"' "-I/home/ubuntu/opt/cores/arduino/avr/cores/arduino" "-I/home/ubuntu/opt/cores/arduino/avr/variants/yun" "/tmp/672686874/build/sketch/exploring_slowLinks.ino.cpp" -o "/dev/null"

Generating function prototypes...

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"' "-I/home/ubuntu/opt/cores/arduino/avr/cores/arduino" "-I/home/ubuntu/opt/cores/arduino/avr/variants/yun" "/tmp/672686874/build/sketch/exploring_slowLinks.ino.cpp" -o "/tmp/672686874/build/preproc/ctags_target_for_gcc_minus_e.cpp"

"/home/ubuntu/opt/arduino-builder/tools/arduino-preprocessor/0.1.5/arduino-preprocessor" "/tmp/672686874/build/preproc/ctags_target_for_gcc_minus_e.cpp" "" -- -std=gnu++11

Compiling sketch...

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_YUN -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8041 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Yun"' "-I/home/ubuntu/opt/cores/arduino/avr/cores/arduino" "-I/home/ubuntu/opt/cores/arduino/avr/variants/yun" "/tmp/672686874/build/sketch/exploring_slowLinks.ino.cpp" -o "/tmp/672686874/build/sketch/exploring_slowLinks.ino.cpp.o"

Compiling libraries...

Compiling core...

Using precompiled core: /tmp/core/core_arduino_avr_yun_86d814ab5ac6e08f9be464f7f3c7beb3.a

Linking everything together...

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega32u4 -o "/tmp/672686874/build/exploring_slowLinks.ino.elf" "/tmp/672686874/build/sketch/exploring_slowLinks.ino.cpp.o" "/tmp/672686874/build/../../core/core_arduino_avr_yun_86d814ab5ac6e08f9be464f7f3c7beb3.a" "-L/tmp/672686874/build" -lm

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "/tmp/672686874/build/exploring_slowLinks.ino.elf" "/tmp/672686874/build/exploring_slowLinks.ino.eep"

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy" -O ihex -R .eeprom "/tmp/672686874/build/exploring_slowLinks.ino.elf" "/tmp/672686874/build/exploring_slowLinks.ino.hex"

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-size" -A "/tmp/672686874/build/exploring_slowLinks.ino.elf"

Sketch uses 3800 bytes (13%) of program storage space. Maximum is 28672 bytes.

Global variables use 163 bytes (6%) of dynamic memory, leaving 2397 bytes for local variables. Maximum is 2560 bytes.

Restarting in bootloader mode

Get port list before reset

[COM1 COM27 COM20 COM9 COM11 COM18 COM17 COM6 COM3 COM29 COM8 COM19]

Touching port COM19 at 1200bps

Set DTR off

Flashing with command:d:/Users/Bob/.arduino-create/arduino/avrdude/6.3.0-arduino9/bin/avrdude.exe -Cd:/Users/Bob/.arduino-create/arduino/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -PCOM19 -b57600 -D -Uflash:w:d:/Users/Bob/AppData/Local/Temp/arduino-create-agent530033685/exploring_slowLinks.hex:i
`

facchinm, 2018-05-04:
Hi @BallscrewBob ,
the output log is missing the error indeed :slightly_smiling_face: 
Could you check if `d:/Users/Bob/.arduino-create/arduino/avrdude/6.3.0-arduino9/bin/avrdude.exe` still exists and has not been deleted/quarantined by any antivirus? If so, please try executing it in a `cmd` shell and report any output, thanks!

BallscrewBob, 2018-05-04:
Will re-install plugin...
Seems odd that it worked for some boards though which is why I didnt suspect that as the issue.

BallscrewBob, 2018-05-04:
Did a proper uninstall (used REVO) and re-install and now It worked on a MEGA but not on a UNO.

Error from UNO upload below.
`
./opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -build-path /tmp/425842614/build -hardware opt/arduino-builder/hardware -hardware ./opt/cores -tools opt/arduino-builder/tools -tools ./opt/tools -built-in-libraries opt/libraries/latest -libraries /tmp/425842614/pinned -libraries /tmp/425842614/custom -fqbn arduino:avr:uno -build-cache /tmp -logger humantags -verbose=true /tmp/425842614/exploring_slowLinks

Using board 'uno' from platform in folder: /home/ubuntu/opt/cores/arduino/avr

Using core 'arduino' from platform in folder: /home/ubuntu/opt/cores/arduino/avr

Detecting libraries used...

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/ubuntu/opt/cores/arduino/avr/cores/arduino" "-I/home/ubuntu/opt/cores/arduino/avr/variants/standard" "/tmp/425842614/build/sketch/exploring_slowLinks.ino.cpp" -o "/dev/null"

Generating function prototypes...

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/ubuntu/opt/cores/arduino/avr/cores/arduino" "-I/home/ubuntu/opt/cores/arduino/avr/variants/standard" "/tmp/425842614/build/sketch/exploring_slowLinks.ino.cpp" -o "/tmp/425842614/build/preproc/ctags_target_for_gcc_minus_e.cpp"

"/home/ubuntu/opt/arduino-builder/tools/arduino-preprocessor/0.1.5/arduino-preprocessor" "/tmp/425842614/build/preproc/ctags_target_for_gcc_minus_e.cpp" "" -- -std=gnu++11

Compiling sketch...

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/ubuntu/opt/cores/arduino/avr/cores/arduino" "-I/home/ubuntu/opt/cores/arduino/avr/variants/standard" "/tmp/425842614/build/sketch/exploring_slowLinks.ino.cpp" -o "/tmp/425842614/build/sketch/exploring_slowLinks.ino.cpp.o"

Compiling libraries...

Compiling core...

Using precompiled core: /tmp/core/core_arduino_avr_uno_86d814ab5ac6e08f9be464f7f3c7beb3.a

Linking everything together...

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o "/tmp/425842614/build/exploring_slowLinks.ino.elf" "/tmp/425842614/build/sketch/exploring_slowLinks.ino.cpp.o" "/tmp/425842614/build/../../core/core_arduino_avr_uno_86d814ab5ac6e08f9be464f7f3c7beb3.a" "-L/tmp/425842614/build" -lm

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "/tmp/425842614/build/exploring_slowLinks.ino.elf" "/tmp/425842614/build/exploring_slowLinks.ino.eep"

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy" -O ihex -R .eeprom "/tmp/425842614/build/exploring_slowLinks.ino.elf" "/tmp/425842614/build/exploring_slowLinks.ino.hex"

"/home/ubuntu/opt/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-size" -A "/tmp/425842614/build/exploring_slowLinks.ino.elf"

Sketch uses 1684 bytes (5%) of program storage space. Maximum is 32256 bytes.

Global variables use 200 bytes (9%) of dynamic memory, leaving 1848 bytes for local variables. Maximum is 2048 bytes.

Flashing with command:d:/Users/Bob/.arduino-create/arduino/avrdude/6.3.0-arduino9/bin/avrdude.exe -Cd:/Users/Bob/.arduino-create/arduino/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega328p -carduino -PCOM20 -b115200 -D -Uflash:w:d:/Users/Bob/AppData/Local/Temp/arduino-create-agent770138261/exploring_slowLinks.hex:i

avrdude: ser_open(): can't open device "\\.\COM20": Access is denied.

avrdude done. Thank you.

`

facchinm, 2018-05-04:
Ok then, the problem is the access denied on `avrdude` , not the tool itself. Is there any other program concurrently accessing the port (like the Java IDE)? Because Mega and UNO share exactly the same uploader, so it makes no sense that one doesn't work if the other does.

BallscrewBob, 2018-05-04:
No the desktop IDE is not open during tests.

I do have two instances of Laser GRBL running but am very specific in not using those two ports.
Will close them down as they have both just finished tasks and try again.

BallscrewBob, 2018-05-04:
More tests needed here but it seems to have helped.

BallscrewBob, 2018-05-04:
Am discounting Laser GRBL as a cause.
Tried a few variations of operating the program but still see the error and also managed to get it to happen to an MKR board. 
It seems a little intermittent and even random as I was able to use the MKR's yesterday.
Not able to nail it down to a single cause !

jawwadturabi, 2019-11-21:
any update over it?

closed 2019-11-21 by BallscrewBob