arduino/arduino-create-agent #237: File access permission issues on plugin update

smellai, 2018-07-02:
Log monitoring is reporting these kind of errors on `/update` request:

```
{"data":{"error":"rename C:\\Users\\[username]\\AppData\\Roaming\\ArduinoCreateAgent-1.1\\.Arduino_Create_Bridge.exe.new C:\\Users\\[username]\\AppData\\Roaming\\ArduinoCreateAgent-1.1\\Arduino_Create_Bridge.exe: The process cannot access the file because it is being used by another process."},"status":"Internal Server Error","method":"POST","browser":"chrome","os":"Windows"}
```
```
{"data":{"error":"open C:\\Users\\[username]\\AppData\\Roaming\\ArduinoCreateAgent-1.1\\Arduino_Create_Bridge.exe: The system cannot find the file specified."},"status":"Internal Server Error","method":"POST","browser":"chrome","os":"Windows"}
```

```
{"data":{"error":"open /home/root/ArduinoCreateAgent-1.1/.Arduino_Create_Bridge.new: permission denied"},"status":"Internal Server Error","method":"POST",,"browser":"chrome","os":"Linux"}
```

```
{"data":{"error":"rename C:\\Users\\[username]\\AppData\\Roaming\\ArduinoCreateAgent-1.1\\Arduino_Create_Bridge.exe C:\\Users\\[username]\\AppData\\Roaming\\ArduinoCreateAgent-1.1\\.Arduino_Create_Bridge.exe.old: Access is denied."},"status":"Internal Server Error","method":"POST","browser":"chrome","os":"Windows"}
```

```
{"data":{"error":"open /home/adl/adl/ArduinoCreateAgent-1.1/.Arduino_Create_Bridge.old: no such file or directory"},"status":"Internal Server Error","method":"POST","browser":"firefox","os":"Linux"}
```

smellai, 2018-07-27:
Happened to me to:
`{"error":"open /home/stefania/.local/share/Trash/files/.Arduino_Create_Bridge_cli.old: no such file or directory"}`

umbynos, 2021-02-09:
The way the update is handled has changed significantly https://github.com/arduino/arduino-create-agent/pull/551
Closing

closed 2021-02-09 by umbynos