arduino/arduino-create-agent #239: Can't connect socket over WSS on firefox

nzaouesgi, 2018-07-20:
Hi everyone !

Here is my problem:

At first I was using **ws://** and then i noticed that Firefox doesn't accept **ws://** connections when the website is over HTTPS (i got the "this is insecure" error from socket.io). Well, my website is over HTTPS.

Changed all my config.ini (origins and signatureKey). I get the endpoints from 127.0.0.1:8991/info but then the socket won't connect, and it reads "**CORS request failure**" from the console.

This is happening **only in Firefox**, Chrome works fine. I tried reinstalling the agent with the certificate several times, even manually, and it _doesn't work_.

Any idea ? Thank a lot.

smellai, 2018-07-23:
in https you should use localhost in place of 127.0.0.1:port

nzaouesgi, 2018-07-23:
Thanks for your answer.

In fact, that's what i did. And it still doesn't work.

mastrolinux, 2018-07-23:
You should also add your domain in the list of origins:

`origins = https://your.domaind.name`

in the config.ini file or it will be not whitelisted.

nzaouesgi, 2018-07-23:
Yes i did that as well.

Firefox says that the certificate issuer is unknown. I know that Firefox doesn't trust Window's certificate store, so that's the reason why. I found the /certificate.crt API in order to make Firefox accept the certificate. 

My question is: is there any other way of doing that ? Like an install script that would modify Firefox built-in certificates DB ? I noticed that there was no need for accessing the /certificate.crt API on the arduino create website, and it works good on Firefox.

Thanks a lot.

smellai, 2018-07-23:
what version of the agent are you using? Actually Firefox should work fine in plain http, using 127.0.0.1 endpoints

nzaouesgi, 2018-07-23:
I'm using version 1.1.71.

The problem happens when i try connecting the websocket. Since the website uses HTTPS, Firefox doesn't allow unsecure websocket connections in **ws://**. Accessing the /info API works fine without HTTPS, but not the socket.io io() function. When i try it in **ws://** (on 127.0.0.1), the connection won't happen, and the console says "The operation is unsecure".

smellai, 2018-07-23:
Do you want to try our new js module? https://github.com/arduino/arduino-create-agent-js-client
Just require 71 version here https://github.com/arduino/arduino-create-agent-js-client/blob/master/src/socket-daemon.js#L31
we are going to release 72 soon

nzaouesgi, 2018-07-23:
Thanks. I already knew about it (looks good btw). If no one has a solution for my issue, then i'll be happy to try it. 👍

mastrolinux, 2018-07-25:
You have to generate the cert by using

`-generateCert` option. Then you can import that cert manually in Firefox so it will work. We do have a script but it is for Linux only.

nzaouesgi, 2018-07-25:
@mastrolinux 

Ok, thanks for the info. Now i'm able to make Firefox trust the CA.

I noticed that using the plugin on the arduino create website doesn't require such procedure. How is this possible ?

mastrolinux, 2018-07-25:
Because we do not use https for localhost because it is completely useless, that's why we us http and 127.0.0.1. https://bugzilla.mozilla.org/show_bug.cgi?id=903966

smellai, 2018-07-25:
Do you have the last version of FF?

mastrolinux, 2018-07-25:
@nzaouesgi you are correct if you use wss that is not possible and you should use https. We do not use wss in our implementation but we do use socket.io that reverts back to long polling in this specific case. So we indeed use long polling, which is not the best solution for performance but still the best from a UX point of view.

nzaouesgi, 2018-07-26:
@mastrolinux thanks, now i get it. issue solved.

@smellai it's all right, mastrolinux solved it !

closed 2018-07-26 by nzaouesgi