arduino/arduino-create-agent #252: Certificates Upon Installation

koussakhan, 2018-08-02:
Hello Arduino Team,

I'm using the **ArduinoCreateAgent-1.1-osx-installer-chrome** installer. When the installation is done, no certificates are being installed, not the Arduino ones nor the localhost ones.
Plus I can only connect to the debug localhost on localhost:8991 and not https://localhost:8992 

The second problem is...
I tried building the agent myself, but no luck. The Arduino certificate is being generated as trusted and the localhost one is being generated but untrusted in the keychain. I need it to be trusted because I automate the installation using the command line command(s): 
`sudo security add-trusted-cert -d -r trustRoot -k "/Library/Keychains/System.keychain" "ca.cert.cer"
`
and
`sudo security add-trusted-cert -d -r trustRoot -k "/Library/Keychains/System.keychain" "cert.cer"
`

Any help would be appreciated, Thank you!

koussakhan, 2018-08-02:
I just rebuilt the **arduino-create-agent** GoLang source-code and modified some text and endpoints in the config.ini file.

Noting down I've set the generateCertificates flag to true before that so I can have some certificates. 

Then I package all the generated certificates with their key.pem(s) and wrote a bash installer. It just moves the files to the needed path and then runs the commands that I've stated above for the certificate installation.

The modified version of the arduino-create-agent which I've made, marks the localhost certificate as untrusted in the Keychain. That is the problem I'm facing. 

Where the original arduino-create-agent (Arduino's), Doesn't add certificates to the keychain. How Am I supposed to communicate using HTTPS websites then?

matteosuppo, 2018-08-02:
@smellai already answered you. I'll quote her for your convenience:

> Hi @koussakhan,
you don't need certificates anymore for chrome as it allows communication to
http://127.0.0.1 endpoints.

smellai, 2018-08-02:
we would really love to know more about your client and the use case in general <3

koussakhan, 2018-08-02:
Hey, sorry if I wrote the message in an unclear way. You stated that no certificates are needed for the http://127.0.0.1:899(1) endpoints, does that include http(s)://127.0.0.1:899(2) too? :)

matteosuppo, 2018-08-02:
If you are in a page residing at an https address, such as `https://example.com`, you can communicate to `http://127.0.0.1:899*` or `ws://127.0.0.1:899*`. You have no need to call the https endpoint.

koussakhan, 2018-08-02:
I thought mixed requests are not possible !! I was able to communicate with the http://127.0.0.1 endpoint using the https platform.

Sorry for the hassle and thank you so much!

closed 2018-08-02 by koussakhan

smellai, 2018-08-02:
They are still not possible in Safari and Edge/IE