arduino/arduino-create-agent #253: Multiple agent instances running on kill / restart

smellai, 2018-08-06:
It seems that killing / restarting the agent doesn't kill it for real, expecially when changing config.ini.

I now have multiple instances running:
[image]

the result is that I have the upload log messages duplicated and serial monitor print too

[image]

See issue report on the forum too:
http://forum.arduino.cc/index.php?topic=550222.0

AlbyIanna, 2018-08-06:
Not happening to me. How are you killing it?

smellai, 2018-08-07:
It seems that duplication of messages is not related to multiple agent instances running, but to multiple socket connections open, I will try to solve it in the frontend, we can keep the issue open for the multiple processes part

smellai, 2020-10-02:
If we could have a way to kill a running agent before starting a new one on the agent side it will be great