arduino/arduino-create-agent #259: signatureKey will not be decoded correctly

tantalate, 2018-09-17:
I am assigning signatureKey as one-line string with `\n` replacing all line breaks, but the `\n` is read as `\\n`  in the signatureKey string, and thus `pem.Decode([]byte(*signatureKey))`  would failed to decode the key.
How should I assign the signatureKey?

facchinm, 2018-09-17:
@tantalate this is quite strange since it used to work exactly as you describe (like here https://github.com/arduino/arduino-create-agent/issues/149#issuecomment-301824017). @matteosuppo any idea?

tantalate, 2018-09-17:
> @tantalate this is quite strange since it used to work exactly as you describe (like here [#149 (comment)](https://github.com/arduino/arduino-create-agent/issues/149#issuecomment-301824017)). @matteosuppo any idea?

Yes, it used to work in 1.1.71, but not working in 1.1.76.

I think it might be introduced by updates of go-ini/ini package.

I have tried to split the key into multiple lines and enclose it in back quotes and it worked.

Thank you!

facchinm, 2018-09-17:
Great! Would you mind adding an example of the config for future reference? And sorry for the regression, we'll notify third party users (OpenRoberta for example) about this.

tantalate, 2018-09-18:
The signatureKey could be set like this:

	signatureKey = `-----BEGIN PUBLIC KEY-----
	MIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAty9EIVIY666Wxst7npCA
	vPAqAjjbSNNk8b3Bkzfs1jtkGYAhXTSI7Cg9oCiuZWXbhAL9+Q9riWxOSZ5r3O1E
	FAVfEIrMFByBSBjyOggUKu/v8XbdWrNHhe4Y6XeJ0cnAygeqL/130szEEbBB3yj+
	AJr+UMLRhBy6Cz7fWBfXaBE0NMLHnlC/yQJsf+PsLclFs2Tqvdq7mL8keZHZGI5p
	0dCjo26RDj+XeT+R/Wmo/rf77stCOcKhdc1pRB9blZkQVemisxN0aX1O4IldkP5A
	iruAyPN3WW06hobsIcGVrPZAznSdsTvlMnrRgJrND0Zut6ZjiqujJxy7AVaRiOr1
	FQIDAQAB
	-----END PUBLIC KEY-----`

dido18, 2025-04-01:
We found a regression on this feature that has been fixed in the agent [1.7.0 ](https://github.com/arduino/arduino-create-agent/releases/tag/1.7.0) 
We have also updated the [wiki](https://github.com/arduino/arduino-create-agent/wiki/Advanced-usage#how-to-use-a-custom-privatepublic-key) page with the instruction to generate and use a custom public/private key.

The format proposed by @tantalate for the `signatureKey` is the only correct format.

closed 2025-04-01 by dido18