arduino/arduino-create-agent #265: device serial number returned only sometimes

smellai, 2018-11-08:
[image]

smellai, 2018-11-08:
See https://github.com/arduino/arduino-create-agent/issues/231

AlbyIanna, 2019-06-04:
As discussed with @cmaglie it appears that this behavior is due to the fact that some boards (mostly new boards) do not expose the serial number when they are in bootloader mode. If you quit the bootloader mode, for example uploading a sketch to the board or clicking the reset button just once, the serial number will be exposed again.
Closing this issue, because it doesn't concern the agent, and needs to be addressed in another way.

closed 2019-06-04 by AlbyIanna