arduino/arduino-create-agent #269: HTTP Endpoint e

Shoodown, 2018-11-26:


smellai, 2018-11-26:
Hi @Shoodown ,
are you using the last version of Firefox?
It could be very interesting for us to know more about your use case and what you're tring to build with our plugin, if you want to share

umbynos, 2021-02-09:
Hi, did you succeed in solving the issue?

mxmarchal, 2021-02-09:
Hi,

It was a long time ago but from what I remember, I started to work on a solution in the source code before giving up because of my lack of knowledge in Go.

So I end up going through another more "homemade" solution using NodeJS inspired by a project I saw on Github but I'm not sure about the name.

I don't have a backup for this project anymore unfortunately.

umbynos, 2021-02-12:
Ok, nice! From my understanding since the ESR problem was solved, we can close this issue. Feel free to reopen it ☺️

closed 2021-02-12 by umbynos