arduino/arduino-create-agent #289: ArduinoCreateAgent not signed resulting in connection blocked by firewall

anwes, 2019-01-28:
Some people reports the [object Object] error in the forums. Checking the log shows the following error:

{"DownloadStatus":"Error","Msg":"Get https://downloads.arduino.cc/packages/package_staging_index.json: dial tcp xxx.xx.xxx.xx:443: connect: host is down"}

Little Snitch on Mac shows that the ArduinoCreateAgent's code signature is not signed and hence created a block rule for all connections to the agent. Overriding the rule solves the [object Object] error.

per1234, 2019-01-28:
Here's the primary forum thread, which has links to the other reports of the issue:
http://forum.arduino.cc/index.php?topic=590962

gvarisco, 2019-01-29:
Thanks for the report, @anwes. We're thinking about a possible fix - I'll get back to you once I have an update.

umbynos, 2021-02-12:
Hi! This issue should be solved in the last version [1.2.1](https://github.com/arduino/arduino-create-agent/releases/tag/1.2.1). Could you please confirm it?

closed 2021-03-08 by github-actions[bot]