arduino/arduino-create-agent #324: Support for additional boards

step21, 2019-03-05:
Is it possible to add support for additional boards? Would this project be the right one for this or which?

umbynos, 2021-02-12:
Hi! Which board are you interested in? This project (as stated in the readme):
>The Arduino Create Agent is a single binary that will sit on the traybar and work in the background. It allows you to use the Arduino Create applications to seamlessly upload code to any USB connected Arduino board (or Yún in LAN) directly from the browser.
Sorry for the delay, are you still interested in this project?

step21, 2021-02-12:
Hey, at the time I think I was mostly interested in esp8266/esp32. It's fine if that is not the intention, I just thought like it is possible to add additional boards the the arduino desktop environment. Or maybe now the new raspberry based arduinos.

umbynos, 2021-02-12:
Actually, we do support ESP 8266 with the [arduino maker plan](https://store.arduino.cc/digital/create). As for the ESP32, it's not possible since the codebase is all python based.
For sure we will add support for the next arduino boards.
Closing now, feel free to reopen it if you need further info :love_letter:

closed 2021-02-12 by umbynos